z88dk/z88dk #1827: sdcc changing its default calling convention

spth, 2021-07-28:
The current default calling convention in SDCC, which passes all parameters on the stack has been criticised by users for a long time. Many feel that using register arguments would be more efficient. Stack cleanup by the callee has also been requested. And when making breaking changes, why not also change the registers used for return values?

Some experiments have been done, and there is now a good idea of how the future default calling convention for SDCC would look like.

Feature request:
https://sourceforge.net/p/sdcc/feature-requests/732/

Branch to implement the new convention:
https://sourceforge.net/p/sdcc/code/HEAD/tree/branches/breaktheworld/

As of today, the branch should be good enough for a bit of first testing for the z80, z180 and z80n (and stm8) ports. Individual functions can be marked as using the old convention via __sdccoldcall.

Maybe z88dk wants to have a look, to see if the new calling convention would work well for it?

feilipu, 2021-07-28:
I'm sitting here, reading the sdcc feature request thinking. OMG this is really breaking the world. And, I don't know whether anyone has capacity to convert all the libraries to a new calling convention. Particularly if it is only to save a handful of percentage points in performance or code size.

Put that aside. When comparing writing libraries for sccz80 and sdcc I do find the stack only convention of sdcc inefficient. For math having the stack and registers is much better. So a change that did nothing else than adopted the sccz80 way of handling one argument (registers) or two arguments (stack and registers) would be great.

At some stage I heard that `bcde` was considered a better set to pass into. But some other existing code uses `hlde`.
Either way, I think it would be good to keep consistency of `bcde`, `de`, and `e`. Or `hlde`, `de`, and `e`.

There is little to be lost for 8-bit single variable functions in moving the argument around. Whether it is in `a`, or in `e`, or in `l` matters not.

All I can add is to _think carefully before changing everything_, when just allowing the first argument to go into registers is sufficient change to deliver most of the improvement. And otherwise just stay with what we've got now `dehl`, `hl`, `l`.

I'm sure there are other more experienced opinions to be heard too.
But that's my 2c.

spth, 2021-07-28:
The idea is that changing the calling convention places a big burden on writers of asm code. So we want to avoid it if possible. We are now at a point where we can't avoid it. But I think it is better to make all the changes we want at once, so we don't have to break the ABI again for the next 10 years or so.
I assume that initially, many people will just add __sdccoldcall to the declarations of their asm functions, to continue using the old calling convention, until they get around to rewriting stuff.

feilipu, 2021-07-28:
> The idea is that changing the calling convention places a big burden on writers of asm code. So we want to avoid it if possible. We are now at a point where we can't avoid it.

But, it seems to be working just fine now... or not?
Don't forget that the silent majority of users are not vocal about change, because they're building on a stable foundation.

Speaking just for z80, z180, and z80n and my math libraries, I've got literally more than a thousand functions that will need to change, each one of which could introduce a bug. And, TBH, for just a 5% improvement it is totally not worth doing. I've got plans for new things that would have to shelved for months while I fix old stuff.

And my math libraries are just a tiny fraction of the z88dk classic library and newlib functions... I don't know whether there is anyone right now who can go through and even apply a decorator to their declarations, and test things aren't broken.

> But I think it is better to make all the changes we want at once, so we don't have to break the ABI again for the next 10 years or so.

My suggestion is first to see what real improvement is available from allowing a register `fastcall` for single argument functions, as a potential change.

And, then with that as a basis measure what changing the existing register usage brings in addition to a register `fastcall`.

Implementing further change for less than 5% increment in either performance or size is just creating unnecessary grind work for everyone who relies on sdcc, imho.

spth, 2021-07-28:
> But, it seems to be working just fine now... or not?

Over the years, SDCC has improved a lot with register allocation, code generation and peephole optimization contributing the most. But the calling convention looks like what could give the most improvement now (though generalized constant propagation would also help, but that is another major task - maybe something for next year).

> Don't forget that the silent majority of users are not vocal about change, because they're building on a stable foundation.

Sure, I'll have to see what comes up on the sdcc-user list, or here.

> Speaking just for z80, z180, and z80n and my math libraries, I've got literally more than a thousand functions that will need to change, each one of which could introduce a bug. And, TBH, for just a [...]

If you feel it isn't worth it for most of your functions, you could just add __sdccoldcall to their declarations. The only thing you'd loose is being able to assign their addresses to function pointers declared without __sdccoldcall.

> My suggestion is first to see what real improvement is available from allowing a register `fastcall` for single argument functions, as a potential change.

I don't remember, but I'll rerurn some experiments, and should be able to report results tonight: current convention vs. new convention vs. just putting the first argument into good registers. For z80, r3ka and gbz80.

If you want me to test a particular choice of registers for the arguments, I could do that, too.
The callingconvention branch is set up to allow easy testing of calling conventions, and I have a Makefile that then also compiles Dhrystone, Whetstone, Coremark and stdcbench with the various calling convention variants.

feilipu, 2021-07-28:
> Stack cleanup by the callee has also been requested.

We have that with the `__z88dk_callee` decoration. It wouldn't be too hard to globally replace that with a generic `__sdcc_callee` decorator. So that's an easy change to support.

> If you want me to test a particular choice of registers for the arguments, I could do that, too.

Perhaps, as we have to deal with 8080 and 8085 too, it would be worth thinking about `dehl` vs `bcde` (which I think is the default from historical code).

To [Sergey's point](https://sourceforge.net/p/sdcc/feature-requests/732/#b871/9d1c), could we keep this breaking change as exclusively opt-in option for a while, at least until a major version increment?

spth, 2021-07-28:
> To [Sergey's point](https://sourceforge.net/p/sdcc/feature-requests/732/#b871/9d1c), could we keep this breaking change as exclusively opt-in option for a while, at least until a major version increment?

For now, the new convention will be in the breaktheworld branch only. I don't think it will be merged to trunk within the next 2 or 3 months. Though maybe the stm8 part will make it to trunk earlier.

spth, 2021-07-29:
All results from callingconvention branch.
[list.gz](https://github.com/z88dk/z88dk/files/6899877/list.gz)

z80:
Dhrystone: old 10688 B, new 10322 B
Whetstone: old 16242 B, new 15115 B
Coremark: old 20553 B, new 20174 B
stdcbench: old 30621 B, new 30022 B

r3ka:
Dhrystone: old 9463 B, new 9097 B
Whetstone: old 13085 B, new 12205 B
Coremark: old 15534 B, new 15343 B
stdcbench: old 25286 B, new 24641 B 

More code size results can be found in the attached list.
Each line looks like:

~~~~
      0	  10342	      0	  10342	   2866	dhrystone-z80-sdcc_args_al_hlde_hlde_ret_a_hl_hlde_callee_00/image.ihx
~~~~

This is the output of the size command for image.ihx. 10343 is the code size in Bytes. This is Dhrystone, compiled using the z80 port. The calling convention passes the first argument in a, if it has 8 bits, hl if it has 16 bits, hlde if it has 32 bits. If the first argument is in a, and hte second has 8 bits, it is passed in l. If the first argument is in a or hl, and the second has 16 bits, it is passed in de. Return values are in 8 for 8 bits, hl for 16 bits, hlde for 32 bits. The stack pointer is always adjusted by the caller (the 00 is a bit mask - if the bit for 1 is set, the callee cleans up the stack for functions returning void, if 2 is set, for functions returning 8 bits, if 4 is set for functions returning 16 bits, the remaining 4 bits are for functions returning 32 bits, depending on the return value being float or not, and on there being a first argument of type float).

Another example:

~~~~
      0	  24549	      0	  24549	   5fe5	stdcbench-r3ka-sdcc_args_2al_2hl_hlde_ret_a_hl_hlde_callee_45/image.ihx
~~~~

24549 B of code size for stdcbench compiled using the r3ka port. First argument in a for 8 bit, hl for 16 bit, hlde for 32 bit. If the first is in a, and the second has 16 bits, it is in hl. If the first is in hl or hlde, and the second has 8 bits, it is in a. return value in a for 8 bits, hl for 16 bits, hlde for 32 bits. Stack cleanupo by caller, with 3 exceptions: Functions that return void, functions that return 16 bits, functions that both return float and have a first argument of type float.

The current calling convention is sdcc_args____ret_l_hl_dehl_callee_00. The new one proposed for z80/z180/z80n is sdcc_args_al_hlde_hlde_ret_a_de_hlde_callee_41, the new one proposed for r3ka/r2ka/r2k/tlcs90/ez80_z80 is sdcc_args_2al_2hl_hlde_ret_a_hl_hlde_callee_41.

feilipu, 2021-07-29:
Ok, it looks like the lines are sorted based on size, right? I see the current calling line here (drystone example).
```
0	  10688	      0	  10688	   29c0	dhrystone-z80-sdcc_args____ret_l_hl_dehl_callee_00/image.ihx
```
And I think the best new option is this, right?
```
0	  10316	      0	  10316	   284c	dhrystone-z80-sdcc_args_al_hlde_hlde_ret_a_de_debc_callee_4b/image.ihx
```
Which is about 3.6% better.

And for stdcbench the current line is this
```
0	  30621	      0	  30621	   779d	stdcbench-z80-sdcc_args____ret_l_hl_dehl_callee_00/image.ihx
```
And the best new option is this...
```
0	  29882	      0	  29882	   74ba	stdcbench-z80-sdcc_args_2ae_2bcde_debc_ret_a_bc_hlde_callee_4c/image.ihx
```
Which is about 2% better.

But, neither of these "best" results are what is being proposed.

The actual proposal is in each case is __3.2%__ better than the current case (total of cycles from all benchmarks).
I think that is pretty clear evidence that SDCC is doing a great job as it is.

The exercise was worth doing to prove that there's  little to be gained from "breakingtheworld".
It should keep the detractors quiet for the next 10 years. :+1:

spth, 2021-07-29:
Well,I suppose there will be discussions with other users, also on the sdcc-user mailing list, so see what various users of the z80 ports thin, before making a decision.
However, even if the default is changed, z88dk could still opt out, as AFAIK z88dk doesn't use the library that comes with SDCC. If the default calling convention is changed in the future, I'd add a --sdccoldcall command line option that makes all functions use the old calling convention by default. This would be a way for z88dk to continue using the old convention without adding __sdccoldcall in many places.
It might also happen that the convention is only changed for some ports, e.g. gbz80 and stm8, while it stays for others.

feilipu, 2021-07-30:
> If the default calling convention is changed in the future, I'd add a --sdccoldcall command line option that makes all functions use the old calling convention by default.

That sounds like a good option, if the z80 calling convention is changed. :+1: 

Unrelated to calling convention, but related to performance improvement. I noticed that the 8051 has assembly floating point solutions integrated (for example [here in `_fsmul.c`](https://sourceforge.net/p/sdcc/code/HEAD/tree/trunk/sdcc/device/lib/_fsmul.c#l34)). If you wanted to make a substantial performance and size improvement for z80, z180 and the SpectrumNext z80n users, then it would be possible to add assembly floating point for those three machines too.

For example with [Whetstone](https://github.com/z88dk/z88dk/wiki/Benchmarks#whetstone-12) on z80, we could do both **30%** reduction in binary size (which is nearly 10x the proposed calling convention improvement) and **4x** the performance! The improvement is even greater for z180 and z80n machines. Let me know if this would be interesting, please.

spth, 2021-07-30:
Regarding floating-point: I opened a ticket in the SDCC feature tracker to discuss the details:https://sourceforge.net/p/sdcc/feature-requests/753/ mcs51 is a special case here, since its asm functions live in the same files as the generic C implementations. Other ports have the asm functions in the respective subdirectories (e.g. z80/abs.s for the implementation of abs() for z80).

spth, 2021-08-03:
Regarding the new calling convention: The breaktheworld branch is now a bit more refined.
There are __sdcccall(0) and __sdcccall(1) for the old and new calling convention. Functions that are not explicitly marked as using one of the two, use the default (currently new convention in that branch, but can be set via --sdcccall 0 or --sdcccall 1 command line option).
The current idea is:
- stm8 and gbz80 go first (i.e. before z80, z80n, z180, ez80_z80, tlcs90, r2k, r3ka, r3ka), since
 a) stm8 and gbz80 are apparently less controversial
 b) stm8 and gbz80 already pass all regression tests in the breaktheworld branch
- support for the new convention gets merged to trunk soon, but the default for now will still be the old convention
- later the default in trunk is switched from --sdcccall 0 to --sdcccall 1, at which time the part of the library implemented in asm will also switch
- Now and while trunk still uses the old convention by default, the breaktheworld branch already uses the new convention by default (and has the asm part of the library ported to the new convention)

spth, 2021-08-20:
As of today, for all of stm8, z80, z80n, z180, gbz80, r2k, r2ka, r3ka, tlcs90, ez80_z80, both breaktheworld and trunk support the new convention. For gbz80 and stm8, regression tests pass in breaktheworld, the other ports still need a bit more work on porting asm-implemented standard library function (for z80, z80n, z180 only the __itoa family hasn't been ported yet). The new conventions are fully documented in the manual.

There are only two differences now between breaktheworld and trunk:

* breakttheworld uses the new convention by default (i.e. in absence of __sdcccall and --sdcccall), while trunk uses the old one by default.
* breakttheworld has the standard library built for the new convention, while trunk has it built for the old convention.

For stm8 and gbz80, trunk will switch to the new convention for default and standard library in the next few weeks, so we get plenty of experience with their new conventions before the next release (likely 4.2.0 in early 2022).

spth, 2021-08-22:
AFAIK, z88dk uses the z80, z80n, z180, ez80_z80, r2ka, r3ka targets of SDCC, and no others. Is that correct?
On the SDCC side we have a plan to make lessen the impact of the transition on those that want to continue to use the old convention, and would likely cover only that targets used by z88dk that way.
The idea is to mark asm-implemented SDCC library function with __sdcccall(1) in the SDCC standard library headers. That way, those that still want the old convention can just
1) Add --sdcccall 0 in the library Makefile before compiling the standard library.
2) Use --sdcccall 0 when compiling their own code.
That way nearly everything would continue to work for them with the old convention, with the exception of some aspects of asm-implemented SDCC standard library functions (assigning them to function pointers - compile time error, accessing them by own declaration instead of standard header - fails silently).

suborb, 2021-08-22:
> AFAIK, z88dk uses the z80, z80n, z180, ez80_z80, r2ka, r3ka targets of SDCC, and no others. Is that correct?

gbz80 is also used.

spth, 2021-08-22:
So all z80-related ports except for r2k and tlcs90 are used?

feilipu, 2021-08-23:
I think that the actual rabbit port used is the older r2k one, rather than r2ka, from [our config file](https://github.com/z88dk/z88dk/blob/master/lib/config/rcmx000.cfg#L15).

The #1580 issue was trying to resolve this, but there are few targets available to encourage follow up.

> P.S.: Does z88dk use (parts of) the standard library that comes with SDCC?

I believe none at all. The [default config line](https://github.com/z88dk/z88dk/blob/master/Makefile#L100) excludes their building.
 Only [the `sdcc` and `sdcpp` binaries](https://github.com/z88dk/z88dk/blob/master/Makefile#L119) are used.

feilipu, 2022-03-08:
@spth do you have a revised timeline or release date in mind for 4.2.0? :thinking: 

There are quite a few issues we are pending close out with #1969, and I  think we'd happy to go with 4.2.0RC2 if you're intending to do the final release more than a few weeks out (post the current political turbulence). Just looking for some guidance.

PR #1969 (2022-03-09): zsdcc - migrate to stable release v4.2.0
Closes #1470
Closes #1827
Closes #1861
Closes #1914
Closes #1967
Closes #1977

Release of upstream [SDCC v4.2.0](https://sourceforge.net/p/sdcc/code/13131/).

__ADDITIONS__

- [x] Need to patch the default calling for all z88dk implemented architectures to `__sdcccall(0)`<br>to ensure no changes are necessary within z88dk libraries.
- [x] Need to provide new intrinsic functions: `___mulsint2slong`,  `___muluint2ulong`.
- [x] Add new intrinsic signed function `___mulsint2slong` to `am9511` math library<br>(however not unsigned, which remains in `l_` libr [... cut]
closed 2022-03-09 by feilipu