z88dk/z88dk #1830: missing files uthash and utarray

gek169, 2021-08-04:
the files uthash and utarray are not found during compilation.

[image]

suborb, 2021-08-04:
I suspect you've not added the submodules. Please take a look at README.md, the relevant snippet is below:

> Use GitHub and build it yourself. The z88dk repository uses git submodules, these are not automatically downloaded by git by default so you will have to either adjust your clone line, or retrieve them manually. To clone with submodules use git clone --recursive https://github.com/z88dk/z88dk.git. To add the submodules to an already existing clone use git submodule update --init --recursive. To build, the following instructions should be followed.

gek169, 2021-08-04:
Ah. Thanks

closed 2021-08-04 by gek169