z88dk/z88dk #1833: z80asm conditional positive call/jump

ssantner, 2021-08-13:
I've been using z80asm and found that it does not assemble the following instructions properly:
"CP label" ---> This is turning into a "CPI" instruction
"JP label" ---> This is turning into a "JMP" instruction

suborb, 2021-08-13:
In z80asm all opcodes are active at all times regardless of the targeted cpu. I think this is a bit of a historical quirk from when just rabbit, z80 and z80n were supported but importantly allows us to use the same assembler source but target different cpus in the libraries. 

Because of the clash the 8080 opcodes ended up being renamed j_p and c_p in z80asm which I know isn’t very useful. 

As a result, I’d recommend using the toz80 awk script in the support folder, or just write z80 in the first place.

feilipu, 2021-08-13:
To help with my thoughts with 8085 code, I edited this [8085 to z80](https://github.com/feilipu/8085-opcodes) mnemonics tool.

Just clone the repo and open in a browser locally.

Z80asm supports those opcodes.

closed 2021-08-21 by feilipu