z88dk/z88dk #1840: z80asm or any optimization in compiling is altering the asm code

mnunesvsc, 2021-08-23:
I wrote a asm code inside a C project, and it works, but one instruction in asm is being translated, for some reason, to a variant code, it works but it somewhat slow can you give-me some light, even an pragma to disable optimization; inside __asm blocks OR any way of writing opcodes directly.

The code, the portion who is altered... 

[code]
.
:
       ex  af,af'     ;// af = 0x0000  '
        pop af         ;// 
        pop iy 
        pop de         ;//

        ;ld  (recuperaOrg1Pat+1),sp      ; grava a origem em recuperaOrg1+1
        ld  hl,sp      ; hl = org 

tileposicionPat:
        ld  bc,(_efdtiles8 + 244)       ; tile 31 of bank 

        push bc        ;// 
        push bc        ;// 
        push bc        ;// 
.
:
[/code]

the generated code becomes this 

`
83b2 ex      de,hl
83b3 ld      de,$0014
83b6 add     hl,de
83b7 ld      ($83d7),hl
83ba pop     ix
83bc pop     de
83bd pop     bc
83be pop     af
83bf pop     hl
83c0 exx     
83c1 ex      af,af'
83c2 pop     af
83c3 pop     iy
83c5 pop     de
=======>

83c6 ld      hl,$0000
83c9 add     hl,sp

== That is the part, it works, but somewhat it affects 'f' on af and the copys goes wrong, maybe some macro, if there is a way to disable the macro only in this part or enable opcodes directly it will be usefull

83ca ld      bc,($deae)
83ce push    bc
83cf push    bc
83d0 push    bc
83d1 push    bc
`

This is the command line to compile 
zcc +zx -vn --opt-code-speed -SO3 -clib=sdcc_iy -startup=31 --max-allocs-per-node3500 @cera.lista -m --list --c-code-in-asm -ocera -create-app -Cz"--sna" 

Thanks in advance.

feilipu, 2021-08-24:
I don't see exactly what is going on, but perhaps some light on what your command line is doing.
 - `--c-code-in-asm` is going to slow down your code, and prevent a large number of optimisations. Typically it will force your code to use caller functions rather than `__z88dk_callee` or `__z88dk_fastcall` parameter passing. This can require the stack management used by the calling function to change substantially. I'd suggest only using it for code flow and basic debugging, but not for code flow tracing. `--list` should be sufficient to see what your code is doing, and won't mess things up.
 - `-SO3` uses the most aggressive peephole set that z88dk has. This peephole set will do speculative instruction reordering, in the hope that following passes of the peephole optimiser will find new optimisation possibilities. But together with C code comments from `--c-code-in-asm` that won't happen. So you end up with a reordered set of assembly, that probably isn't faster.
 - `--max-allocs-per-node3500` is on the low side for good code. The SDCC libraries are built with 25,000 by default for instance. When testing some of my code won't build with less than 10,000, and for final builds I'm using more like 200,000 or 400,000.

I suggest to start and your `__asm` `__endasm` code block with an assembly comment. This will block the peephole optimiser from reordering with other generated assembly. You should be able to find your assembly code block untouched in the listing.

Otherwise, please post an excerpt of the C together with your asm section, and the resulting listing (without using `--c-code-in-asm`) so we can see what's going on.

mnunesvsc, 2021-08-24:
Thanks for the fast response, indeed this is my command for debug, i use for understand the C code assembled, my production command is the one above

zcc +zx -vn --opt-code-speed -SO3 -clib=sdcc_iy -startup=31 --max-allocs-per-node400000 @cera.lista -oceraprd -s --list -create-app 

but the code does the same, can you give me an sample of that asm  assembly comment. is this just an line with;

;comment... 

or something else ?

feilipu, 2021-08-24:
> Is this just an line with;

> ;comment...

Yes, exactly.

Try an assembly comment before and after to prevent peephole optimisation from seeing your code.

feilipu, 2021-08-24:
Thinking further.

Also try enclosing in do{}while code block.

Like [this](https://github.com/feilipu/z88dk-libraries/blob/master/freertos/source/include/sdcc/portmacro.h) using example for sdcc.

mnunesvsc, 2021-08-24:
Thanks for your time, i did and the result follows;

The code, original, just part it is extense 
`
// Copy entire screen from a similar buffer directly to vmem
void copyScreenYosPopPat(unsigned char *src, unsigned char pat)
{
    // declare parameters 
    src, pat;    

    __asm
        ;comment... to prevent peephole 
		;di
        ld (copyScreenYosPopEndPat+1),sp   ;// store stack pointer.
...
... 

copyScreenYosPopEndPat:
		ld sp,0 
		;ei
        ;end comment... to prevent peephole 

    __endasm;    
}
`

the lis code, generated by z88dk compilation , debug version 
`
1088  0230  C1          	pop	bc
1089  0231  F1          	pop	af
1090  0232  D9          	exx
1091  0233  08          	ex	af,af'
1092  0234  E1          	pop	hl
1093  0235  F1          	pop	af
1094  0236  FD E1       	pop	iy
1095  0238  D1          	pop	de
1096  0239              ;ld	(recuperaOrg1Pat+1),sp ; grava a origem em recuperaOrg1+1
1097  0239  D9          	exx
1098  023A  21 00 00 39 	ld	hl,sp
1099  023E  D9          	exx
1100  023F              	tileposicionPat:
1101  023F  ED 4B 2C 01 	ld	bc,(_efdtiles8 + 300)
`

And a screen of the emulated sna, cant save the text so send a pic 
[image]

Thanks for the help

mnunesvsc, 2021-08-24:
Sorry forgeted the code 

`
// Copy entire screen from a similar buffer directly to vmem
void copyScreenYosPopPat(unsigned char *src, unsigned char pat)
{
// declare parameters
src, pat;

__asm
    ;comment... to prevent peephole 
	;di
    ld (copyScreenYosPopEndPat+1),sp   ;// store stack pointer.

...................................

      ex  af,af'     ;// af = 0x0000  '
      pop hl         ;// 
      pop af         ;// 
      pop iy 
      pop de         ;//    
      exx 
      ld  hl,sp      ; hl = org 
      exx 

............................

      copyScreenYosPopEndPat:
      ld sp,0
      ;ei
      ;end comment... to prevent peephole
      
      __endasm;    
      }
`

the lis code,  generated by z88dk,  z80asm  code generator 

``
`1088  0230  C1          	pop	bc
1089  0231  F1          	pop	af
1090  0232  D9          	exx
1091  0233  08          	ex	af,af'
1092  0234  E1          	pop	hl
1093  0235  F1          	pop	af
1094  0236  FD E1       	pop	iy
1095  0238  D1          	pop	de
1096  0239              ;ld	(recuperaOrg1Pat+1),sp ; grava a origem em recuperaOrg1+1
1097  0239  D9          	exx
1098  023A  21 00 00 39 	ld	hl,sp
1099  023E  D9          	exx
1100  023F              	tileposicionPat:
1101  023F  ED 4B 2C 01 	ld	bc,(_efdtiles8 + 300)
`

feilipu, 2021-08-25:
I am seeing the code of interest as,

```
      ex  af,af'     ;// af = 0x0000  '
      pop hl         ;// 
      pop af         ;// 
      pop iy 
      pop de         ;//    
      exx 
      ld  hl,sp      ; hl = org 
      exx
```

and the listing generated as,
```
1091  0233  08          	ex	af,af'
1092  0234  E1          	pop	hl
1093  0235  F1          	pop	af
1094  0236  FD E1       	pop	iy
1095  0238  D1          	pop	de
1096  0239              ;ld	(recuperaOrg1Pat+1),sp ; grava a origem em recuperaOrg1+1
1097  0239  D9          	exx
1098  023A  21 00 00 39 	ld	hl,sp
1099  023E  D9          	exx
```
Which looks perfectly ok.  So where is the problem?

I don't see the assembly comments in the listing, so I assume you're not sending the whole function or the whole listing?

What's going on in these sections?
```
............................
```

mnunesvsc, 2021-08-25:
Well that is my problem what is happening is some macro or whatswhoever is transforming 

**ld  hl,sp  opcode F9** 

in something else, normally 

**ld, hl,0x0000
add al, sp** 

which is a way to put **sp** in **hl** but is slower, expensive, and mess **f** flags but in the same code this do not occur 

and on an emulator the code transforms into that as show in the image.

[image]

thanks for your attention.

feilipu, 2021-08-25:
They are two different things.

Opcode __F9h__ refers to __`ld sp,hl`__, which puts `hl` into `sp`.

The macro instruction __`ld hl,sp`__ is realised by two instructions `ld hl,00` and `add hl,sp`
There is no Z80 instruction to put `sp` into `hl` directly, I'm afraid.
What you're seeing is the best way to do it.

On 8085 there's another better way, but that won't work for a Z80.

mnunesvsc, 2021-08-25:
Well it is weird, because on the same code i had an **F9** and it works, as the picture of the emulator shows.
[image]

and on this page ( **https://clrhome.org/table/**) it says the F9 is a opcode of Z80, i think it is not the problem.
[image]

my problem is, somewhat the compiler are generating an code, not wrong, but no desirable, and i am asking how can i overcome that.

thanks in advance.

ped7g, 2021-08-25:
> because on the same code i had an **F9** and it works, as the picture of the emulator shows.

the picture shows `ld sp,hl` Also the clrhome table shows `ld sp,hl`. I don't see any `ld hl,sp`