z88dk/z88dk #1849: (newlib) Remove +vgl target

suborb, 2021-08-31:
Because:

* It's not being used
* It's not being maintained
* Classic has +gl which supports more models and has more features

feilipu, 2021-09-01:
[Information](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/vgl/_INFORMATION) that would be worth preserving somewhere as examples, if the +vgl target is removed.

@hotkeymuc would be able to comment on whether the classic +gl target is sufficient, as z88dk classic and new targets are becoming similar.

hotkeymuc, 2021-09-13:
The gl target seems to have become a sufficient replacement for the vgl target as far as I can tell.

@feilipu The additional resources you have [linked](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/target/vgl/_INFORMATION) are worth keeping - albeit in a Wiki rather than in the source repository itself. It contains, for example, code for the bit-banged "soft serial port". I guess that things like that should be put in a separate library than be part of the target.

The lack of (my) maintenance comes from the fact that I have created a dedicated [VTech Genius Leader SDK on Github](https://github.com/hotkeymuc/vgldk) in the meantime, where I dive deeper into the different models. I am happy to augment the z88dk gl target with the things I've learned since, including a 19200 baud serial implementation, LCD graphics, and mouse support for later models.

closed 2021-09-20 by suborb

feilipu, 2021-09-21:
This https://github.com/z88dk/z88dk/blob/master/lib/vgl_crt0.opt
and this https://github.com/z88dk/z88dk/blob/master/lib/vgl_crt0.opt
should go too.