z88dk/z88dk #1856: (z80asm) Can't generate labels with > 255 characters

suborb, 2021-09-10:
The debugging work that's going uses defc constants to encode the CDB debug information.

Unfortunately it looks like when we start to instrument structures, the length could exceed 255 fairly easily.

When this happens we end up with the fatal message: "too long for byte counted string".

@pauloscustodio is there any chance we could shift to word counted strings?

pauloscustodio, 2021-09-14:
Work started; will replace all byte-counted strings in the object files by word-counted strings. Inside z80asm all identifiers are handled as allocated strings, i.e. there is no fixed length limit.

desertkun, 2021-09-14:
C-can we consider ditching url encoding in favor for base64 encoding?

suborb, 2021-09-14:
We'd still need support for labels > 255 characters (for large structs) so the encoding mechanism doesn't matter massively

commit a825e053 (2021-09-14): Fix #1856: add tests
commit 4c0b9d17 (2021-09-14): Fix #1856: store names in object file as word-counted string instead of byte-counted string
commit 0d06c1f9 (2021-09-15): Merge pull request #1858 from z88dk/issue1856

Fix #1856 - store strings in object file as word-counted strings
closed 2021-09-15 by pauloscustodio