z88dk/z88dk #1859: Which version of the Specrtum Next CPU is targeted by z88dk?

spth, 2021-09-17:
SDCC has a z80n port targeting the Z80N cpu in the ZX Spectrum Next.
Currently, the target architecture if v1 of the Z80N. SDCC is considering changing this to v2 (which means SDCC could emit the bsla, bsra, bsrl, srrf and brlc instructions).

Would this cause problems for z88dk?

Philipp

ped7g, 2021-09-17:
I'm not z88dk contributor or user, so my comment is from "outside" (I develop for Next in assembly only).

At this moment there's near zero value to limit yourself to v1, the current widely spread Next core is 3.1.5 (and big update is incoming, probably in couple of months with shipment of second kickstarter units - with no changes in z80n, but some new stuff like timers/etc..), and if some user would post any issue with core 1.x, they would be just asked to upgrade (even anyone with core 2.x or 3.0 would be just asked to upgrade) - and I haven't seen any such post for long time, I would assume all active Next users are now at 3.1.5 or testing the beta of newer cores.

suborb, 2021-09-19:
Nope, shouldn't be a problem if the syntax used by asxxxx is sensible

closed 2021-09-19 by suborb