z88dk/z88dk #1865: z80asm - emulated indirect opcodes

feilipu, 2021-09-25:
From [here](https://github.com/z88dk/z88dk/pull/1863#discussion_r716020514).

As far as I can see `ld de,(**)` is not emulated for INTEL or GBZ80.
It throws a warning, and I can't otherwise find it.

`ld (**),de` would be a useful partner to be emulated for INTEL or GBZ80.
And one more `ld (de),*` to add for all CPUs to complete the set using `ex de,hl`.

@pauloscustodio Could it (they) be added to z80asm, please?

pauloscustodio, 2021-09-25:
gbz80 lacks the `ld hl,(**)` and `ld (**),hl`; the only way to emulate would be through `a`

feilipu, 2021-09-25:
Just for INTEL would suit me.

GBZ80 was just nice to have.

pauloscustodio, 2021-09-25:
gbz80 also lacks `ex de,hl`, therefore `ld r,(de)` / `ld (de),r` / `ld (de),N` only implemented for the other CPUs

closed 2021-09-25 by pauloscustodio

feilipu, 2021-09-26:
Thanks :+1:

commit d8e378be (2021-09-25): Fix #1865 - emulate de-indirect loads

Emulate `ld (de),r`; `ld r,(de)`, `ld (de),N`; for all CPUs except gbz80, as this one lacks `ex de,hl`
Emulate `ld (NN),de`; `ld de,(NN)` for Intel CPUs; gbz80 lacks `ld hl,(NN)` and `ld (NN),hl`