z88dk/z88dk #1866: [cpc][msx] border and background color setting

Fabrizio-Caruso, 2021-09-26:
It seems that border color setting is not implemented on some targets, e.g. MSX and CPC.

Moreover background color setting does not produce a color (after a clear screen) on the CPC target.
It looks like the clear screen is not properly filling the screen with white spaces and that is why background color + clear screen does not produce a colored background.
I am assuming that textbackground followed by clear screen should color the background. Is this correct?

I can try to reproduce with small examples.

My code is doing something like:
bordercolor(WHITE); -> no effect on CPC and MSX
textbackground(WHITE); -> OK on MSX, KO on CPC (

[image]

[image]

Fabrizio-Caruso, 2021-09-27:
@suborb is this issue related to #1867? 
Maybe the border issue is the same but on the CPC I also see a problem with the background.

suborb, 2021-09-27:
It's a fairly easy fix in the libraries to sort out the background - just give me a few days and I'll be able to do it.

suborb, 2021-09-28:
Fixed by:: 79c034c11d6a0729e79659617a79105b2bc4a3fc

closed 2021-09-28 by suborb

Fabrizio-Caruso, 2021-09-30:
@suborb, your fix is ok for the border but not ok for the background because it uses a wrong color (blue instead of white)

[image]

Fabrizio-Caruso, 2021-10-01:
@suborb The fix only fixes the border. The foreground looks better (no stripes) but it has a totally wrong color (blue instead of white):
[image]