z88dk/z88dk #1868: [linux] How to install Z88DK under Linux? The standard procedure sort of fails but I see binaries.

Fabrizio-Caruso, 2021-09-28:
I am trying to install Z88DK (latest version) under Ubuntu (inside the Windows Subsystem).

I have followed:
https://github.com/z88dk/z88dk/wiki/installation#linux--unix

I have omitted the ZSDCC setting to make it simpler.

The compilation fails with
```
...
gcc -o z88dk-lib  z88dk-lib.o
install z88dk-lib /home/fcaruso/z88dk/bin/z88dk-lib
make[1]: Leaving directory '/home/fcaruso/z88dk/src/z88dk-lib'
make -C src/zx0 PREFIX=`pwd` install
make[1]: Entering directory '/home/fcaruso/z88dk/src/zx0'
gcc -g -O2   -c -o compress.o compress.c
gcc -g -O2   -c -o optimize.o optimize.c
gcc -g -O2   -c -o zx0.o zx0.c
gcc -g -O2   -c -o memory.o memory.c
gcc -o z88dk-zx0  compress.o  optimize.o  zx0.o memory.o
gcc -g -O2   -c -o dzx0.o dzx0.c
gcc -o z88dk-dzx0  dzx0.o
install z88dk-zx0 /home/fcaruso/z88dk/bin/z88dk-zx0
install z88dk-dzx0 /home/fcaruso/z88dk/bin/z88dk-dzx0
make[1]: Leaving directory '/home/fcaruso/z88dk/src/zx0'
make -C testsuite
make[1]: Entering directory '/home/fcaruso/z88dk/testsuite'
zcc +test -vn -a Issue_896_nested_struct_initialisation.c -o Issue_896_nested_struct_initialisation.opt.tmp1
make[1]: execvp: zcc: Permission denied
make[1]: *** [Makefile:47: Issue_896_nested_struct_initialisation.opt] Error 127
make[1]: Leaving directory '/home/fcaruso/z88dk/testsuite'
make: *** [Makefile:221: testsuite] Error 2
```

But I see binaries. What may be missing?

suborb, 2021-09-28:
Odd.

What are the permissions on the binaries?

Fabrizio-Caruso, 2021-09-28:
@suborb I have managed to use it, anyway.
It seems only the tests fail. This might be related to how the Windows subsystem works but it would be nice if the installation process worked 100% out of the box.

By the way I am now testing the whole Cross-Lib framework under Linux..
It is now a full framework with automatically generated code and assets... also for user-defined games.

feilipu, 2021-09-29:
I notice the tests fail on my machine too.

But I assumed it was either a) something related to the perl submodules, or b) some cruft in my tree.
Which tests fail exactly?

feilipu, 2021-10-01:
The issue for me is a perl module.
```
make[1]: Leaving directory '/home/phillip/Z80/z88dk/src/common'
make -C ../src/z80asm test
make[1]: Entering directory '/home/phillip/Z80/z88dk/src/z80asm'
make -C dev/z80asm_lib
make[2]: Entering directory '/home/phillip/Z80/z88dk/src/z80asm/dev/z80asm_lib'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/phillip/Z80/z88dk/src/z80asm/dev/z80asm_lib'
cp dev/z80asm_lib/z88dk-z80asm-*.lib .
cp dev/z80asm_lib/z88dk-z80asm-*.lib /home/phillip/Z80/z88dk/lib/config//../
make dev
make[2]: Entering directory '/home/phillip/Z80/z88dk/src/z80asm'
perl asmstyle.pl dev/z80asm_lib/*.asm
Can't locate Modern/Perl.pm in @INC (you may need to install the Modern::Perl module) (@INC contains: /etc/perl /usr/local/lib/x86_64-linux-gnu/perl/5.30.0 /usr/local/share/perl/5.30.0 /usr/lib/x86_64-linux-gnu/perl5/5.30 /usr/share/perl5 /usr/lib/x86_64-linux-gnu/perl/5.30 /usr/share/perl/5.30 /usr/local/lib/site_perl /usr/lib/x86_64-linux-gnu/perl-base) at asmstyle.pl line 10.
BEGIN failed--compilation aborted at asmstyle.pl line 10.
make[2]: *** [Makefile:111: dev] Error 2
make[2]: Leaving directory '/home/phillip/Z80/z88dk/src/z80asm'
make[1]: *** [Makefile:169: test] Error 2
make[1]: Leaving directory '/home/phillip/Z80/z88dk/src/z80asm'
make: *** [Makefile:30: src-all] Error 2
make: Leaving directory '/home/phillip/Z80/z88dk/test'
```
And the issue was that the ownership of the `~/.cpan/` directory was assigned to `root` not to the user (me).

Probably not your issue.

Fabrizio-Caruso, 2021-10-02:
I have now tried the same on a fresh VM and it worked with no issue.

closed 2021-10-02 by Fabrizio-Caruso