z88dk/z88dk #1869: [SCCZ80] Recent bug `assertion failed in modlink.c:512`. It was OK a couple of days ago.

Fabrizio-Caruso, 2021-09-29:
This is a very recent bug. It was not present in the snapshot from 18/09/2021.
It is present in today's snapshot (29/09/2021).

If I try to compile the "tiles" test in Cross-Lib for the Aquarius using SCCZ80 I get this error:

```
-o ../build/Xtiles_aquarius_exp_16k
assertion failed in modlink.c:512
make: *** [tests/tiles/Makefile.tiles:691: aquarius_exp_16k] Error 1
````

In Cross-Lib this can be reproduced from the `src` folder by running:
`xl build tiles aquarius`
which is equivalent to
`make aquarius -f tests/tiles/Makefile.tiles`

The Makefile target looks like
```
aquarius_exp_16k: 
	$(Z88DK_PATH)$(MYZ88DK) +aquarius $(SCCZ80_OPTS) -clib=ansi $(INCLUDE_OPTS) -vn \
	-DSOUNDS -D__AQUARIUS__ \
	-lndos $(NO_UDG_FLAG) \
	-D_XL_SLOW_DOWN_FACTOR=$(_COMMON_SLOWDOWN_FACTOR)*$(_AQUARIUS_SLOWDOWN) \
	-DCONIO -DNO_INIT_GRAPHICS \
	$(FULL_FILES) \
	-create-app	\
	-o $(BUILD_PATH)/X$(GAME_NAME)_aquarius_exp_16k
```

The other games aned tests seem to compile OK.

suborb, 2021-09-29:
Ah, that's an interesting one.

There's a circular reference that has been created by the the jp reduction stuff I've been doing lately:

```
        defc    i_64 = i_61
        defc    i_61 = i_64
```

I think it must be being caused by a `while(1){}` or similar loop at the end of main - I'll try and figure out a way to solve it

@pauloscustodio - it's an odd problem, but is there any way to get a better diagnostic than just the assertion?

pauloscustodio, 2021-09-30:
I will have a look to the assertion failure

commit c18190d5 (2021-09-30): Fix #1869: detect and report symbols not defined because of circular dependencies

insead of assertion failure
commit e12622c6 (2021-09-30): Merge pull request #1872 from z88dk/issue1869

Fix #1869: detect and report symbols not defined because of circular …
closed 2021-09-30 by pauloscustodio