z88dk/z88dk #1870: [question] Can I distribute freely a docker image with a fully set-up Cross-Lib enviroment including Z88DK?

Fabrizio-Caruso, 2021-09-29:
Cross-Lib depends on 
more than a dozen cross-compilers 
https://github.com/Fabrizio-Caruso/CROSS-LIB/blob/master/docs/COMPILERS.md, 
a native compiler, make, python and few other tools.

Could I create a docker image with a fully set-up Cross-Lib environment (including Z88DK) and distribute it freely?
Obviously I would include links to the Z88DK and all the other compilers.
Cross-Lib is a totally free (for non-commercial use) and open source project.

This would make Cross-Lib so much easier to use for other people.

suborb, 2021-09-30:
I'd be very wary about producing a docker image with all the required tool chains - it will be huge and a maintenance issue

An image per tool chain is probably a more scalable way to do things and also allows you to delegate docker image maintenance to the upstream projects.

z88dk has a docker file, the usage is documented here: https://github.com/z88dk/z88dk/wiki/Docker-Usage - previously we were producing the images nightly but dockerhub being dockerhub stopped. I'm also not convinced there was a massive number of users.

Fabrizio-Caruso, 2021-09-30:
@suborb, Thanks for your advice! 
My current goal is much more modest.

The problem is that Cross-Lib depends on many cross-compilers. 
I will always maintain Cross-Lib as a stand-alone library without the compilers but at the same time I want to make it simpler for casual people who want to give it a try.
My idea is to have a docker image with stable compiler versions that are tested to compile all my games and tests.
I won't maintain the docker image. I will updated it once in a long while. 
The best way to use Cross-Lib would still be to use it as now, i.e. as a separate library in a properly manually set-up environment.
I am working on documenting how to set it up under Windows and under Linux and I am improving support for Linux.

Fabrizio-Caruso, 2021-09-30:
@zx70 Yes, I am asking for permission but I am also happy to receive any advice, comments, suggestions, critics, etc.

Installing a fully setup Cross-Lib environment for someone who just wants to give it a try, appears like a daunting task.
I want to make it simple for people who want to try it. This frozen stable version, should be packed with all tools and compilers and just work out of the box and compile all my games and tests for all 200 targets.
Getting a version of each compiler (mainly CC65 and Z88DK) to work on hundreds of builds is something I would like to spare to the casual user.

On the other end, docker images (once I figure out how to build them) won't be my standard releases. I will release Cross-Lib as a separate tool. I am working on ways to make it easy to use on both Windows+Cygwin and plain Linux.

desertkun, 2021-09-30:
Having a docker container image would be nice. In absence of one, [I am forced to build one myself](https://github.com/the-channels/channels/blob/master/Dockerfile#L8).

Fabrizio-Caruso, 2021-09-30:
@derekfountain I am not talking about a docker image for Z88DK.
I am talking about a docker image for a Cross-Lib  (https://github.com/Fabrizio-Caruso/CROSS-LIB) set-up, i.e.:
- Cross-Lib (https://github.com/Fabrizio-Caruso/CROSS-LIB)
- potentially more than a dozen compilers (including native and cross GCC variants, CC65, Z88DK, CMOC and others)
- some tools (make, python, java, perl, etc.)

Cross-Lib is a WORA (write once, run anywhere) framework to code universal 8-bit games with the *very same* code on about 200 vintage targets over all vintage 8-bit architectures. So it depends on many cross-compilers.
Cross-Lib is free (for non-commercial use) and open source. If I pack it inside CC65, Z88DK, etc... I may need to ask for permission.