z88dk/z88dk #1877: [MSX] Compilation with '-debug' option fails in case of included 'stdio.h'

masushin, 2021-10-03:
For following test code ,

```
#include <msx.h>
#include <msx/gfx.h>
#include <msx/line.h>

void main() {
    set_mode(1);
    set_color(15,4,4);
    set_mode(2);
    line(0,0,100,100);
    line(100,0,0,100);
}
```

Compile with '-debug' and run it.
```
$ zcc +msx -subtype=rom -create-app -debug -o test.rom test.c
Program requires cartridge with 1 16k ROM banks
Notice: Available RAM space is 16144 bytes ignoring the stack
Adding main bank 0x00 (14290 bytes free)

$ openmsx test.rom
```

this is work correctly,
Crossed lines will be appeared.

But, for the following which is added stdio.h,

```
#include <msx.h>
#include <msx/gfx.h>
#include <msx/line.h>

#include <stdio.h> //  <-- add

void main() {
    set_mode(1);
    set_color(15,4,4);
    set_mode(2);
    line(0,0,100,100);
    line(100,0,0,100);
}
```

Compilation will fails like the following.
```
$ zcc +msx -subtype=rom -create-app -debug -o test.rom test.c
Error at file 'test.c::main::1::2' line 14: syntax error
Error at file 'test.c::main::1::2' line 14: syntax error
Error at file 'test.c::main::1::2' line 14: symbol '__CDBINFO__T_3aF_2fopt_2fz_38_38dk_2flib_2fconfig_2f_2e_2e_2f_2e_2e_2f_2finclude_2fcpm_2eh_24fcb_5b_28_7b_30_7dS_3aS_24drive_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_7dS_3aS_24name_24_30_5f_30_24_30_28_7b_38_7dDA_38x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_39_7dS_3aS_24ext_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_32_7dS_3aS_24extent_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_33_7dS_3aS_24filler_24_30_5f_30_24_30_28_7b_32_7dDA_32x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_35_7dS_3aS_24records_24_30_5f_30_24_30_28_7b_31_7dSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_36_7dS_3aS_24discmap_24_30_5f_30_24_30_28_7b_31_36_7dDA_31_36x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_32_7dS_3aS_24next_5frecord_24_30_5f_30_24_30_28_7b_31_7dSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_33_7dS_3aS_24ranrec_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_33_36_7dS_3aS_24rwptr_24_30_5f_30_24_30_28_7b_34_7dSL_3aU_29_2cZ_2c_30_2c_30_29_2' not defined
Error at file 'test.c::main::1::2' line 14: symbol '__CDBINFO__T_3aF_2fopt_2fz_38_38dk_2flib_2fconfig_2f_2e_2e_2f_2e_2e_2f_2finclude_2fcpm_2eh_24sfcb_5b_28_7b_30_7dS_3aS_24drive_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_7dS_3aS_24name_24_30_5f_30_24_30_28_7b_38_7dDA_38x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_39_7dS_3aS_24ext_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_32_7dS_3aS_24pwdmode_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_33_7dS_3aS_24filler_24_30_5f_30_24_30_28_7b_31_31_7dDA_31_31x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_32_34_7dS_3aS_24c_5fdate_24_30_5f_30_24_30_28_7b_32_7dSI_3aS_29_2cZ_2c_30_2c_30_29_28_7b_32_36_7dS_3aS_24c_5fhours_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_32_37_7dS_3aS_24c_5fminutes_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_32_38_7dS_3aS_24date_24_30_5f_30_24_30_28_7b_32_7dSI_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_30_7dS_3aS_24hours_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_33_31_7dS_3a' not defined
Errors in source file test.c:
Error at file 'test.c::main::1::2' line 14: syntax error
                   ^ ---- }
Error at file 'test.c::main::1::2' line 14: syntax error
                   ^ ---- }
Error at file 'test.c::main::1::2' line 14: symbol '__CDBINFO__T_3aF_2fopt_2fz_38_38dk_2flib_2fconfig_2f_2e_2e_2f_2e_2e_2f_2finclude_2fcpm_2eh_24fcb_5b_28_7b_30_7dS_3aS_24drive_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_7dS_3aS_24name_24_30_5f_30_24_30_28_7b_38_7dDA_38x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_39_7dS_3aS_24ext_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_32_7dS_3aS_24extent_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_33_7dS_3aS_24filler_24_30_5f_30_24_30_28_7b_32_7dDA_32x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_35_7dS_3aS_24records_24_30_5f_30_24_30_28_7b_31_7dSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_36_7dS_3aS_24discmap_24_30_5f_30_24_30_28_7b_31_36_7dDA_31_36x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_32_7dS_3aS_24next_5frecord_24_30_5f_30_24_30_28_7b_31_7dSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_33_7dS_3aS_24ranrec_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_33_36_7dS_3aS_24rwptr_24_30_5f_30_24_                   ^ ---- }
30_28_7b_34_7dSL_3aU_29_2cZ_2c_30_2c_30_29_2' not defined
Error at file 'test.c::main::1::2' line 14: symbol '__CDBINFO__T_3aF_2fopt_2fz_38_38dk_2flib_2fconfig_2f_2e_2e_2f_2e_2e_2f_2finclude_2fcpm_2eh_24sfcb_5b_28_7b_30_7dS_3aS_24drive_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_7dS_3aS_24name_24_30_5f_30_24_30_28_7b_38_7dDA_38x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_39_7dS_3aS_24ext_24_30_5f_30_24_30_28_7b_33_7dDA_33x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_31_32_7dS_3aS_24pwdmode_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_31_33_7dS_3aS_24filler_24_30_5f_30_24_30_28_7b_31_31_7dDA_31_31x_2cSC_3aS_29_2cZ_2c_30_2c_30_29_28_7b_32_34_7dS_3aS_24c_5fdate_24_30_5f_30_24_30_28_7b_32_7dSI_3aS_29_2cZ_2c_30_2c_30_29_28_7b_32_36_7dS_3aS_24c_5fhours_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_32_37_7dS_3aS_24c_5fminutes_24_30_5f_30_24_30_28_7b_31_7dSC_3aU_29_2cZ_2c_30_2c_30_29_28_7b_32_38_7dS_3aS_24date_24_30_5f_30_24_30_28_7b_32_7dSI_3aS_29_2cZ_2c_30_2c_30_29_28_7b_33_30_7dS_3aS_24hours_24_30_5f_30_24_30_28_7b_31_7dSC_                   ^ ---- }
3aU_29_2cZ_2c_30_2c_30_29_28_7b_33_31_7dS_3a' not defined
```

What is the cause of this ?
Is there a workaround for this ?

suborb, 2021-10-03:
That looks like a debug line for a struct definition. 

Regardless of the fact that ticks hasn’t been updated yet it shouldn’t cause a problem. 

I’ll investigate - just to confirm this the latest nightly?

masushin, 2021-10-03:
Ok, I'll prepare the latest nightly build environment and confirm it.

suborb, 2021-10-03:
It's still present in nightly, looks like a line length somewhere isn't quite big enough to hold the full cdb object.

closed 2021-10-03 by suborb

masushin, 2021-10-04:
@suborb 
Let me confirm.

https://github.com/z88dk/z88dk/issues/1856

Is it mean that the above is not related to this problem, and this is caused by my environment?

suborb, 2021-10-04:
The problem was that the debug information for a struct definition was being truncated. 

I’ve now made the buffer larger so it shouldn’t happen and compilation should work.

masushin, 2021-10-05:
Oh ... I missed that you fixed it.
Thank you very much.

commit 8e9777a4 (2021-10-03): Fix #1877: 32k is big enough for everyone right?

CDB info can exceed 1k, so bump the size up a little