z88dk/z88dk #1878: Mac M1 Support

egruttner, 2021-10-04:
Hi,
I'm getting this error when I try to run zcc:

zsh: bad CPU type in executable: ./bin/zcc

I'm using a Macbook Pro M1 - macOS Big Sur 11.6
Is the M1 supported?

Thank you in advance.

suborb, 2021-10-04:
The binaries produced by nightly are meant to be Intel 64 bit binaries:

```
file zcc
zcc: Mach-O 64-bit x86_64 executable, flags:<NOUNDEFS|DYLDLINK|TWOLEVEL>
```

Rosetta 2 should be kicking in, you may need to install this manually:

```
softwareupdate --install-rosetta
```

I'm afraid I don't have access to M1 mac for testing.

egruttner, 2021-10-04:
Thank you, the problem only occurs with 1.99C and previous versions

Tested OK with 2.0 and 2.1

closed 2021-10-04 by egruttner

suborb, 2021-10-04:
Ah right. Yes 1.99C contained 32 bit Intel binaries.

suborb, 2022-09-27:
From 20220928 nightly builds should be a fat binary with arm64 and x86_64 architectures.

For my next trick I'll try to figure out the code signing...

desertkun, 2022-09-27:
If you're not looking into shelling $100 I've got an active dev account. If that helps.

suborb, 2022-09-27:
Well, today I think it's about £80, but tomorrow who knows....

I've got the actual signing process working with a free developer key, but obviously that fails `spctl assess` on a different machine. 

If you don't mind emailing me a "Developer ID Application" p12 then I can verify that things work and then decide...

desertkun, 2022-09-28:
Sent 👍