z88dk/z88dk #1879: sccz80 - atof return in alt registers

feilipu, 2021-10-07:
When returning from `asm_atof` ([`asm_strtod`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/stdlib/z80/asm_strtod.asm)) with a float (both 4 byte and 6 byte) it is expected the returned float is in the alternate registers.

sccz80 stores this float using `dstore` or `l_plong` for 6 byte floats or 4 byte floats respectively.

For 6 byte floats it is correctly stored from exx with `dstore`. But for 4 byte floats using `l_plong` there is a missing `exx` as `l_plong` expects that the long is in the front registers.

suborb, 2021-10-07:
My gut says that `asm_strtod()` is or rather `atof()/strtod()` is broken and returning incorrectly.

A 32bit float is treated as a int32 by both sdcc and sccz80 so should be in the main set.

How is sdcc working?

feilipu, 2021-10-07:
> This would need to be done for all library routines that return a float/double.

:scream:

feilipu, 2021-10-07:
Behaviour notes...

sdcc does the same thing, irrespective of the (math) library, because it only handles 32-bit floats, assuming `atof` returns to front registers.

sccz80 doesn't exclusively call a library function to store the returned float. Sometimes it will store 4 bytes directly (incorrectly assuming that it is in the front registers).

But if the sccz80 library `atof`always returns to front registers, then it would be wrong for 6 byte maths libraries using `dstore`.

So does the glue function for `atof` have to differentiate between 4 and 6 byte library variants? This is a bit messy.

I'm not sure whether you've come to the same conclusion, but I think this has to be standardised inside sccz80 as either front or back register return expectation, and then the libraries adjusted to match. :thinking: 

`zcc +cpm -SO3 --list -clib=sdcc_iy  startrek.c -o startrek -lm -create-app`

```
1576  0385  CD 00 00    	call	_atof_fastcall
1577  0388  4D          	ld	c, l
1578  0389  DD 6E FD    	ld	l,(ix-3)
1579  038C  44          	ld	b,h
1580  038D  DD 66 FE    	ld	h,(ix-2)
1581  0390  E3          	ex	(sp), hl
```
`zcc +cpm -SO3 --list -clib=sdcc_iy  startrek.c -o startrek --math32 -create-app`

```
1594  0385  CD 00 00    	call	_atof_fastcall
1595  0388  4D          	ld	c, l
1596  0389  DD 6E FD    	ld	l,(ix-3)
1597  038C  44          	ld	b,h
1598  038D  DD 66 FE    	ld	h,(ix-2)
1599  0390  E3          	ex	(sp), hl
```
zcc +cpm -SO3 --list startrek.c -o startrek --math32 -create-app -compiler=sdcc

```
1493  0392  CD 00 00    	call	_atof
1494  0395  F1          	pop	af
1495  0396  C1          	pop	bc
1496  0397  DD 75 FC    	ld	(ix-4),l
1497  039A  DD 74 FD    	ld	(ix-3),h
1498  039D  DD 73 FE    	ld	(ix-2),e
1499  03A0  DD 72 FF    	ld	(ix-1),d
```

sccz80 does different things, depending on 4 or 6 byte floats.

`zcc +cpm -SO3 --list -clib=new  startrek.c -o startrek -lm -create-app`

```
366   0470  CD 00 00    	call	atof
366   0473  E1          	pop	hl
366   0474  CD 00 00    	call	dstore

366   054D  CD 00 00    	call	atof
366   0550  21 F4 01    	ld	hl,_w1
366   0553  CD 00 00    	call	dstore
```

sccz80 will load directly (no call) if it sees the option to do this.

`zcc +cpm -SO3 --list -clib=new  startrek.c -o startrek --math32 -create-app`

```
366   046D  CD 00 00    	call	atof
366   0470  C1          	pop	bc
366   0471  CD 00 00    	call	l_plong

366   054C  CD 00 00    	call	atof
366   054F  22 D6 01    	ld	(_w1),hl
366   0552  ED 53 D8 01 	ld	(_w1+2),de
```
`zcc +cpm -SO3 --list  startrek.c -o startrek -lm -create-app`

```
366   047F  CD 00 00    	call	atof
366   0482  C1          	pop	bc
366   0483  E1          	pop	hl
366   0484  CD 00 00    	call	dstore

366   055E  CD 00 00    	call	atof
366   0561  C1          	pop	bc
366   0562  21 F4 01    	ld	hl,_w1
366   0565  CD 00 00    	call	dstore
```

Here sccz80 doesn't call a library function to store the float, but assumes that it will be returned in the front registers.

`zcc +cpm -SO3 --list  startrek.c -o startrek --math32 -create-app`

```
366   055D  CD 00 00    	call	atof
366   0560  C1          	pop	bc
366   0561  22 D6 01    	ld	(_w1),hl
366   0564  ED 53 D8 01 	ld	(_w1+2),de

366   055D  CD 00 00    	call	atof
366   0560  C1          	pop	bc
366   0561  22 D6 01    	ld	(_w1),hl
366   0564  ED 53 D8 01 	ld	(_w1+2),de
```

feilipu, 2021-10-08:
And Lt. Sulu was made happy.

suborb, 2021-10-08:
Nice work, that one has been there for a while!