z88dk/z88dk #1886: [SCCZ80][-O3][bug][all targets] A very recent change is producing broken binaries

Fabrizio-Caruso, 2021-10-10:
The newest version of Z88DK with -O3 produces broken binaries. -O2 seems OK.

For example my game Cross Horde (see screenshot below) is now broken: moving to the left leaves traces of the bow on all Z88DK (SCCZ80) versions.

It looks like the problem is that some code is skipped, e.g., the code that is supposed to remove traces of the moving bow in my game.

I can see that this is a very recent bug as it was ok on the 4th of October.
For example:
04/10/2021 OK
09/10/2021 KO

I am compiling with:
```
msx_32k_rom: $(ASSETS_PATH)/z88dk_xchase.asm   
	$(Z88DK_PATH)$(MYZ88DK) +msx $(GLOBAL_OPTS) -O3 $(INCLUDE_OPTS) \
	-pragma-redirect:fputc_cons=fputc_cons_generic \
...
	-vn   -D__MSX__  \
	-lndos -subtype=rom \
	-create-app -o $(BUILD_PATH)/X$(GAME_NAME)_msx_32k.rom \
...
	$(ASSETS_PATH)/z88dk_xchase.asm \
	$(CROSS_LIB_PATH)/display/init_graphics/z88dk/udg/udg_init_graphics.c \
	$(CROSS_LIB_PATH)/sound/z88dk/psg/ay-3-8910_psg_sounds.c \
	$(FULL_FILES)
	rm $(BUILD_PATH)/X$(GAME_NAME)_msx_32k_BSS.bin
	rm $(BUILD_PATH)/X$(GAME_NAME)_msx_32k_DATA.bin
```

[image]

Fabrizio-Caruso, 2021-10-10:
`xl horde msx` -> Builds Cross Horde for the msx
`xl horde` -> Builds Cross Horde for the native terminal (useful to test without an emulator)
`xl horde z88dk_targets` -> Builds Cross Horde for all supported default z88dk targets
`xl horde cc65_targets` -> same for cc65 targets
`xl horde cmoc_targets` -> same for cmoc targets
etc.
`xl all z88dk_targets` -> Builds all games and tests for all z88dk_targets

suborb, 2021-10-10:
I actually can't reproduce this, the bow is cleared correctly - could you attach the ROM and a .map file for the defective build?

Fabrizio-Caruso, 2021-10-10:
@suborb the bow gets corrupted only when you move to the left.

[build.zip](https://github.com/z88dk/z88dk/files/7317303/build.zip)

suborb, 2021-10-10:
I can with your build but not with mine. The bow movement is really smooth - for me. I guess it's still UDGs so it's really nice!

Lets see what's different though..

Fabrizio-Caruso, 2021-10-10:
I have the same issue with the snapshot from last night (2021/10/10). 
I am testing with an old emulator (BlueMSX).

suborb, 2021-10-10:
Apologies, I've managed to reproduce it.

It looks like it's a rules issue and an interaction between what's done at O2 and what's done at O3.

Fabrizio-Caruso, 2021-10-10:
@suborb Any suggestion on how to have my Makefiles work on all POSIX targets is welcome.
I know how to have my makefile behave differently between Windows/Cygwin and Linux propoer but how can I detect MacOS and BSD?
If the find with parameter also works on Windows/Cygwin and Linux, then I could just use that version.

Can I test it on MacOS (possibly legally) on a virtual machine?

Fabrizio-Caruso, 2021-10-10:
Thanks! I have updated the find command line.

By the way, may I ask you if you have tried the script `xl` to perform builds, create projects, delete projects, etc. ??
If yes, did you work for you (except delete and reset, which require the "find" fix) ?

pauloscustodio, 2021-10-10:
> find -type f -name '*.o' -delete

You can use:
`find . -type f -name '*.o' -exec rm {} \;`

it works in MacOS and Linux.

suborb, 2021-10-10:
```
./xl
zsh: ./xl: bad interpreter: /bin/python: no such file or directory
```

Change the shebang to this:

```
#!/usr/bin/env python
```