z88dk/z88dk #1898: Request for ASSERT directive for z80asm

dbolli, 2021-12-04:
Other assemblers have an ASSERT (evaluated expression) e.g. (constant1 <= constant2) that throws an error if the assert fails which is useful e.g. for ensuring that code does not cross an address boundary.

Would be very useful if z80asm had this feature.

pauloscustodio, 2021-12-25:
@dbolli thanks for the suggestion

closed 2021-12-25 by pauloscustodio

dbolli, 2021-12-26:
Works like a charm. Many thanks :)

defc ORGADDR = $6100

ASSERT ( ASMPC + $6100 > DFILE )			               # :dbolli:20211227 10:58:55 Works for pre-defined constant
ASSERT ( $ + $6100 >= START )			                  # :dbolli:20211227 10:58:55 Works for pre-defined label
ASSERT ( $ + ORGADDR <= $8000 )			               # :dbolli:20211227 10:58:55 Works for pre-defined constant

commit 3993a259 (2021-12-25): Fix #1898: add ASSERT directive to abort compilation if expression is false
commit 858422dd (2021-12-25): Merge pull request #1915 from z88dk/issue1898

Fix #1898: add ASSERT directive to abort compilation if expression is…