z88dk/z88dk #1899: Can't compile with the "-compiler=sdcc" option

arcturus-prime, 2021-12-04:
Not sure what's up, but I took a look and found a similar issue here already (see #1399). This is specifically what I'm getting:
`user@machine:~$ zcc +ti8x -compiler=sdcc test.c`
`sh: 1: z88dk-zsdcpp: not found`

Nothing else comes up. When attempting to run these commands manually, `z88dk-zsdcpp` doesn't exist, only `z88dk-zcpp` does. I never changed any of the default build options, and to my knowledge SDCC is installed by default. Any ideas?

arcturus-prime, 2021-12-04:
Update, I just retried the same command with `zsdcc` and that worked. The documentation seems to suggest that `sdcc` is a valid compiler option. Is this just not the case?

feilipu, 2021-12-05:
The hint is in this line.

z88dk-zsdcpp: not found

You need to ensure zsdcc and zsdcc are installed and are properly named.

The nightly for Windows has these. For git clones you'll need to build them and move/rename them yourself.

Follow the recipe in the top level docker file or make file. And/or read and follow the installation instructions.

arcturus-prime, 2021-12-05:
Ah, I missed the part about exporting the BUILD_SDCC flag before building the Linux nightly. I didn't do that. Apologies for wasting your time, my bad. I know dealing with, frankly, stupid GitHub issues created by users is irritating. I suppose I should have read more carefully.
I'm still somewhat perplexed though, why did `-compiler=zsdcc` work then? Does it select sccz80 by default if an invalid compiler option is supplied?

suborb, 2021-12-05:
> I'm still somewhat perplexed though, why did `-compiler=zsdcc` work then? Does it select sccz80 by default if an invalid compiler option is supplied?

I think it might do which is bad. I shall make it grumble instead.

commit ae4fd463 (2021-12-12): Fix #1899: Abort for an unknown compiler
closed 2021-12-12 by suborb