z88dk/z88dk #1908: sccz80 compile errors with -m8085

feilipu, 2021-12-17:
I'm compiling the z88dk-lib fatfs code for the 8085 CPU.

From the [source directory of ff](https://github.com/feilipu/z88dk-libraries/tree/master/ff/source), issuing this command generates a correct z80 library.
```
zcc +rc2014 -mz80 -x -O2 @ff.lst -o ../ff
```

But this command for 8085 generates a sea of error messages.
```
zcc +rc2014 -m8085 -x -O2 @ff.lst -o ../ff
```
Not even sure where to start to fix this...

suborb, 2021-12-17:
Hmmm, I may not have all the right changes, but:

```
zcc +rc2014 -clib=new -m8085 @ff.lst -x -o ../ff
```

Seems to work - note the -clib=new to switch to using sccz80 as the compiler.

```
zcc +rc2014 -m8085 @ff.lst -x -o ../ff
```

Which will use sdcc generates a lot of errors since the sdcc code will use ix which isn't available on the 8085.

feilipu, 2021-12-17:
Yes. That works. But I thought that would generate a newlib library.

But I was trying to get a classic library. Is it incorrect to assume the no option default is sccz80/classic ?

Or is there no such thing as a classic external library?

feilipu, 2021-12-17:
Just thinking further, the target definition configures the default compiler/library. Hence the need to explicitly restate sccz80 as the desired compiler.

feilipu, 2021-12-17:
Works as suggested.
Thanks. :+1:

closed 2021-12-17 by feilipu