z88dk/z88dk #1909: sccz80 - 8080/8085 - missing __builtin_*

feilipu, 2021-12-19:
[Code generation for 8080 and 8085](https://github.com/z88dk/z88dk/blob/7e4f6dd5b3be21875c02bbe21e4fe9eff45a8ea4/src/sccz80/callfunc.c#L352) seems to expect the presence of below functions.
```
GLOBAL	__builtin_memset
GLOBAL	__builtin_memcpy
GLOBAL	__builtin_strcpy
GLOBAL	__builtin_strchr
```
The [codegen.c](https://github.com/z88dk/z88dk/blob/master/src/sccz80/codegen.c#L4993) doesn't seem to generate 8080/8085 code properly, so this code isn't used.
Rather sccz80 is relying on the builtin functions, that aren't provided currently.

Not sure of the best way to fix this?
- Add the missing linkage to (for example) `memset` or `memset_callee`?
- Or change sccz80 to produce the right inline code?

suborb, 2021-12-19:
I suspect inlining wouldn't save much, so I'm happy to not have builtins for 808x. Indeed, it should be setup this way, so I think you've just got a missing -D somewhere:

They're defined here:
 
https://github.com/z88dk/z88dk/blob/master/include/string.h#L402

and usually disabled by this bit here:

https://github.com/z88dk/z88dk/blob/master/include/sys/compiler.h#L30

feilipu, 2021-12-19:
It is odd that including the new [`sys/compiler.h`](https://github.com/z88dk/z88dk/commit/936df596fef8ee323cdebf640f43af955e6dfc4b) file doesn't seem to prevent sccz80 from generating the `__builtin_*` functions.

The `-D__DISABLE_BUILTIN` command line option seems to be necessary to stop the calls to `__builtin_*` from being generated.

commit 936df596 (2021-12-19): newlib - add sys/compiler.h closes #1909
closed 2021-12-19 by feilipu