z88dk/z88dk #1910: classic - 8085 - amalloc not working

feilipu, 2021-12-19:
I think I have broken something with the 8085 (and others) implementation of amalloc (or indeed it may not ever have been truly working correctly). The culprit will probably be in the [crt clib variables set up](https://github.com/z88dk/z88dk/blame/master/libsrc/_DEVELOPMENT/target/clib_variables.inc), since that is the piece I touched last.

Just noting this here in case others are noticing the issue too.

feilipu, 2021-12-19:
Hmm. Wonder if there's an issue in my setup which uses classic AMALLOC doing the setup using `sbrk_callee`, but the program is calling `malloc()`, and the newlib malloc is initialised by `asm_heap_init` ?

feilipu, 2021-12-20:
@suborb I wonder could you help with this, please?
I don't understand where I'm going wrong.

There is an issue with returning an allocation from classic malloc. It always returns a null value.

The classic library process for heap allocation seems to be done correctly in my crt code, and the [classic malloc](https://github.com/z88dk/z88dk/blob/master/libsrc/alloc/malloc-classic/malloc.asm) function is being used by the library. But somewhere there is a mismatch.

I've attached a listing from a file built using the [acia85 subtype](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/startup/rc2014_crt_2.asm.m4) (subtype 2), showing the startup process. [rc2014-acia85.lis](https://github.com/z88dk/z88dk/files/7743059/rc2014-acia85.txt)

suborb, 2021-12-20:
Quick pre-breakfast glance - I’ll look in depth later. 

The assembly for ld hl,sp looks a bit off? 4 bytes is coming out.

feilipu, 2021-12-20:
> Quick pre-breakfast glance - I’ll look in depth later.

Thanks. No rush.
I can work around the issue with static arrays in the interim. So its not a blocker.

suborb, 2021-12-20:
Testing malloc with +test works fine, so it must be something in your setup - I can reproduce it which is going to be very handy!

```
#include <stdio.h>
#include <malloc.h>

int main() {
        void *p = malloc(100);
        printf("%p\n",p);
        while(1);
}
```

```
% zcc +rc2014 -subtype=acia85 alloc.c -m -DAMALLOC -create-app
% z88dk-ticks -m8085 a.rom
0
```

feilipu, 2021-12-21:
> Prepare to kick yourself...

Thanks. Much appreciated. Duly kicked.

I'd looked at the ordering and was suspicious. But the comment gave false security. https://github.com/z88dk/z88dk/commit/470dc87a52129ed2836c7b2d886476071aeca8e7

It is fixed now. https://github.com/z88dk/z88dk/commit/fc464a98085f15f97a47f1d33ec44676165a102f

closed 2021-12-21 by feilipu