z88dk/z88dk #1911: intel / gbz80 - optimise copy / set loops

feilipu, 2021-12-22:
For the Intel / GBZ80 newlib/classic implementations the implementation of copy / set loops could be improved.

eg. [asm_memcpy](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_memcpy.asm#L86), and [asm_memset](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_memset.asm#L82).

In z80asm [_z80asm_ldir](https://github.com/z88dk/z88dk/blob/master/src/z80asm/dev/z80asm_lib/ldir.asm#L10).

The copy loop currently looks like:
```asm
.asm_ldir
    ;
.loop
    ; stuff to copy / set etc
    dec bc        ; 6
    ld a,b        ; 4
    or c          ; 4
    jr NZ,loop    ; 10
```
Making 24 cycles per byte or total __6144__ cycles per page copied.

An alternative implementation looks like:
```asm
.asm_ldir
    ld a,c         ; 4
    dec bc         ; 6
    inc b          ; 4
    ld c,a         ; 4
.loop
    ; stuff to copy / set etc
    dec c          ; 4
    jr NZ,loop     ; 10
    dec b          ; 4
    jr NZ,loop     ; 10
```
making 14 cycles per byte or __3616__ cycles per page (or less) copied.

@suborb If you're ok with the change, I can do a PR across the library functions.

But @pauloscustodio, would you be able to do the same for z80asm please?
Not sure I can get all the build tests etc right.

ped7g, 2021-12-22:
the prologue can be:
```
  dec bc  ; 6
  inc b  ; 4
  inc c  ; 4
```
-4T and doesn't clobber A

suborb, 2021-12-22:
@feilipu go for it

pauloscustodio, 2021-12-22:
This is a space/time tradeoff. The proposed solution is faster but costs 5 more bytes.

Maybe we can keep two versions of the library functions (e.g. asm_ldir and asm_ldir_fast), call the shorter routine by default and call the faster routine if the z80asm option -opt-speed is passed (this option today only replaces JR by JP).

To build the whole system with the option without changing the build scripts, the Z80ASM environment variable can be used, e.g.
```
Z80ASM=-opt-speed make
```

feilipu, 2021-12-22:
> This is a space/time tradeoff. The proposed solution is faster but costs 5 more bytes.

True, but it is a "shared" 5 bytes, used everywhere where`ldir` (and friends) are called, which may be often in one program.
The outcome is saving half the loop overhead, which for (eg a rewritten :star: ) `asm_memset` would be majority of the cost.
5 bytes per function well spent.

IMHO, you'd be making life complicated for yourself to put in a switch for fast/small here.
It is in a different category of `jr`/`jp` which costs more every time a jump is used.

:star: `asm_memset` can be rewritten to use the `a` register to set `(hl)` rather than copying like `asm_memcpy`.
This will speed it up even more and save some bytes too.

feilipu, 2021-12-23:
With #1912 I've finished the library parts of this, as far as I think worth doing.
Some of the `asm_str*` functions are tested in the suite, but I'm not sure if the `asm_mem*` functions are tested by the build test suite at all.

Takes just under 1800 ticks off each of the string test totals, which isn't bad for these short string tests.
I've added in some [missing GBZ80 routines](https://github.com/z88dk/z88dk/pull/1912/files#diff-977fc1fd9986529592512a169cc71d6e665b9ad7062473c9f499130faf97a3c6L17) and [tests](https://github.com/z88dk/z88dk/pull/1912/files#diff-e29a405c8c2d873bc712c666d5c9e1881e5893399426653c6e8570d48de24eb2L8), so that's why the ticks is higher than previously.

__OLD__
```
Starting 8080 test 
Ticks: 1799023

Starting 8085 test 
Ticks: 1756693

Starting gbz80 test 
Ticks: 2212356
```

__NEW__
```
Starting 8080 test
Ticks: 1797224

Starting 8085 test 
Ticks: 1754926

Starting gbz80 test 
Ticks: 2219076
```

pauloscustodio, 2021-12-25:
Updated ldir and lddr loops in the z80asm emulation library. Thanks @feilipu

feilipu, 2021-12-27:
@suborb I think some of the sccz80 string C linkages (particularly the callee versions) are not INTEL or GBZ80 safe.
I'll go through and make them INTEL safe (no `pop af` `push af`), but not change broken GBZ80 stuff.
eg. [_memstrcpy__callee](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/c/sccz80/_memstrcpy__callee.asm)

On a second pass I'll try to make them GBZ80 safe too (additionally no `ex (sp),hl`).

That's the right thing to do, right?

suborb, 2021-12-27:
Check the header - I think I disabled callee for quite a few functions because of that problem.

feilipu, 2021-12-27:
Ok. I'll do what I can then, and re-enable those that are done.

feilipu, 2021-12-27:
__ORIGINAL__
```
Starting 8080 test 
Ticks: 1799023

Starting 8085 test 
Ticks: 1756693

Starting gbz80 test 
Ticks: 2212356
```

__ASM REWORK__
```
Starting 8080 test
Ticks: 1797224

Starting 8085 test 
Ticks: 1754926

Starting gbz80 test 
Ticks: 2219076 (added new functions & tests)
```
Reworked the sccz80 C linkage for 8080 and 8085 allowing callee for most (all) functions. https://github.com/z88dk/z88dk/commit/b60612292332f6cafa991ee35084345f5c3aa9fe

__C REWORK__
```
Starting 8080 test
Ticks: 1795597

Starting 8085 test 
Ticks: 1753252

Starting gbz80 test 
Ticks: 2219076 (unchanged)
```

closed 2021-12-27 by feilipu