z88dk/z88dk #1913: z88dk-dis does not include labels in listing from map/sym file that are not in library source code e.g. ROM CALLs

dbolli, 2021-12-24:
The .map file generated by z80asm contains
STACKA                          = $2D28 ; const, local, , asm-lib-test, asm_lib_test, ../z88dk-zxspectrum-equates.asm:61
STACKBC                         = $2D2B ; const, local, , asm-lib-test, asm_lib_test, ../z88dk-zxspectrum-equates.asm:62
FPTOBC                          = $2DA2 ; const, local, , asm-lib-test, asm_lib_test, ../z88dk-zxspectrum-equates.asm:63
but the .lis file generated by
z88dk-dis -o $(($ORGADDRESS)) -x $Z80EXECNAME.map $Z80EXECNAME.bin > $Z80EXECNAME-asm.tmp
tr '[:lower:]' '[:upper:]' < $Z80EXECNAME-asm.tmp > $Z80EXECNAME-asm.lis
picks up the labels in library code but not the ROM CALLs
                    LD      BC,$1234                        ;[61E5] 01 34 12
                    CALL    $2D2B                           ;[61E8] CD 2B 2D
                    LD      BC,$3456                        ;[61EB] 01 56 34
                    CALL    $2D2B                           ;[61EE] CD 2B 2D
                    LD      A,$78                           ;[61F1] 3E 78
                    CALL    $2D28                           ;[61F3] CD 28 2D
                    CALL    CALCSTACKTICKER                 ;[61F6] CD CB 73
                    CALL    $2DA2                           ;[61F9] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[61FC] CD CB 73
                    CALL    $2DA2                           ;[61FF] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[6202] CD CB 73
                    CALL    $2DA2                           ;[6205] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[6208] CD CB 73

suborb, 2021-12-27:
I think that's the eternal internal quandary regarding label lookup - at the moment it deliberately doesn't match const values, only address types.

If you change the `SYM_ADDRESS` to `SYM_ANY` inside `src/ticks/disassembler_alg.c` then I think it will do as you expect.

Ideas are very welcome to make it a bit smarter: I'm trying to avoid picking up the internally generated z80asm entries but it's probably skipping too many useful labels as well.

dbolli, 2021-12-27:
@suborb Thanks for your response, making that change now works for constants

                    LD      BC,$1234                        ;[61E5] 01 34 12
                    CALL    STACKBC                         ;[61E8] CD 2B 2D
                    LD      BC,$3456                        ;[61EB] 01 56 34
                    CALL    STACKBC                         ;[61EE] CD 2B 2D
                    LD      A,$78                           ;[61F1] 3E 78
                    CALL    STACKA                          ;[61F3] CD 28 2D
                    CALL    CALCSTACKTICKER                 ;[61F6] CD CB 73
                    CALL    FPTOBC                          ;[61F9] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[61FC] CD CB 73
                    CALL    FPTOBC                          ;[61FF] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[6202] CD CB 73
                    CALL    FPTOBC                          ;[6205] CD A2 2D
                    CALL    CALCSTACKTICKER                 ;[6208] CD CB 73

I notice that local labels don't get picked up
e.g. in .map file
PLRMASK1                        = $62AE ; addr, local, , asm-lib-test, asm_lib_data, asm-lib-test.asm:315
outputs
                    LD      HL,$62AE                        ;[6154] 21 AE 62
instead of
                    LD      HL,PLRMASK1                 ;[6154] 21 AE 62
as from what I can see it doesn't look for labels via find_symbol() for those opcodes.

Maybe a future enhancement could be adding find_symbol() calls and controlling the scope via a command line option?