z88dk/z88dk #1914: SDCC strange issue with this declaration

nippur72, 2021-12-24:
SDCC, in the below code, if I move the `data` declaration (the one marked with *****) to the top of the function, then the function behaves incorrectly. I was not able to understand the reason, all I see is that the resulting `.lis` file has different register allocations. Perhaps it has something to do with optimizations? I've tried to remove the `-SO3` flag but the result was the same. It works normally in sccz80.

```c
void gr_erasepiece(sprite *p) {
   int px = p->x;
   int py = p->y;

   // are we erasing the "next" piece ?
   if(py==PIECE_IS_NEXT) {
      px = NEXT_X;
      py = NEXT_Y;
   }
   else {
      px += STARTBOARD_X;
      py += STARTBOARD_Y;
   }

   // ***********************************
   tile_offset *data = get_piece_offsets(p->piece, p->angle);   

   SLOT1_VIDEO_START();
   for(byte t=0; t<4; t++) {
      int x = px + data->offset_x;
      int y = py + data->offset_y;      
      data++;
      gr4_tile((byte)x,(byte)y,EMPTY_GR_CHAR,EMPTY_GR_COLOR,FONTS);
   }
   SLOT1_END();
}

typedef struct {
   byte offset_x;
   byte offset_y;
} tile_offset;

tile_offset *get_piece_offsets(byte piece, byte angle) {
   return &pieces_XY[(piece*sizeof(tile_offset)*4*2)+angle*4];
}
```

suborb, 2021-12-29:
Sorry, I've not had a chance to investigate this one yet.

Could you add the definitions for sprite and the various preprocessor `#define`s?

nippur72, 2021-12-30:
if it's easier for you to reproduce it, my repo has the whole source code (tetris game):

1) clone https://github.com/nippur72/laser500emu
2) the directory `software\tetris` contains the program that I compile with:
```
zcc +laser500 tetris500.c -o tetris500.bin -I..\lib500 -create-app -Cz--audio --list -m -s -DSDCC -compiler sdcc -SO3
```
the issue is located in `grboard.h`, uncomment line 159 and comment line 174.

I checked it again with latest night build `v19053-73554631a-20211230` and the issue is still there. If you play the game on my Laser 500 emulator you'll see dropping tetris pieces acting oddly (drag and drop the compiled `tetris500.bin` over the [emulator window](https://nippur72.github.io/laser500emu).

suborb, 2021-12-30:
It looks like this is easiest to see from the mangled "next up" box.

I've gone back through some old sdcc versions, and the last working one I can find is r12069 which was in nightly builds from 22/4 to 29/4 this year.

There's been a fair amount of churn and hence codegen bugs in sdcc this year so I think the next thing to try is to upgrade: we're already pending one bug fix so let me try that.

If that doesn't solve it, then this issue will be difficult to report upstream: an issue which just uses prints the wrong values is easier to figure out.

suborb, 2021-12-30:
Unfortunately, I can still reproduce the mangled "next up" box with the latest version of sdcc (r12831)

feilipu, 2022-02-13:
Confirmed that the issue is still present with r13006.
Mangled "up next" box, and other overwrite issues.

[image]

feilipu, 2022-02-15:
Added [SDCC issue 3322](https://sourceforge.net/p/sdcc/bugs/3322/).

PR #1969 (2022-03-09): zsdcc - migrate to stable release v4.2.0
Closes #1470
Closes #1827
Closes #1861
Closes #1914
Closes #1967
Closes #1977

Release of upstream [SDCC v4.2.0](https://sourceforge.net/p/sdcc/code/13131/).

__ADDITIONS__

- [x] Need to patch the default calling for all z88dk implemented architectures to `__sdcccall(0)`<br>to ensure no changes are necessary within z88dk libraries.
- [x] Need to provide new intrinsic functions: `___mulsint2slong`,  `___muluint2ulong`.
- [x] Add new intrinsic signed function `___mulsint2slong` to `am9511` math library<br>(however not unsigned, which remains in `l_` libr [... cut]
closed 2022-03-09 by feilipu