z88dk/z88dk #1916: classic - override library PUBLIC declarations

feilipu, 2021-12-28:
When building a project I'd like to override certain `PUBLIC` declarations in the library device and driver files with my own project specific declarations. This override process is working with sccz80 (and sdcc) in the newlib, but I don't seem to get  the same result with my classic library based 8085 build.

Specifically, the `PUBLIC` declaration of the device locations (eg [ACIA Tx Buffer here](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/rc2014/device/acia/8080/__acia_data_Tx.asm)) need to be overridden.
It works with the [newlib command line](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE#building) for the [`acia` version](https://github.com/RC2014Z80/RC2014/tree/master/ROMs/CPM-IDE/acia).
I've not pushed the `acia85` version yet (as it is not working), but it differs only in the 8085 specifics unrelated to this issue.

``` sh
zcc +rc2014 -subtype=acia -clib=new -O2 -m --math32 -llib/rc2014/ff_ro @cpm22.lst -o ../rc2014-acia-cpm22 -create-app
zcc +rc2014 -subtype=acia85 -O2 -m -D__CLASSIC -DAMALLOC --math32 -l_DEVELOPMENT/lib/sccz80/lib/rc2014/ff_85_ro @cpm22.lst -o ../rc2014-acia85-cpm22 -create-app
```
I'm getting already "defined errors" when the linker is building my acia85 project version.
Perhaps this is something to do with the order the libraries are linked?

``` txt
Error at module 'cpm22_asm': symbol 'aciaRxCount' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaRxIn' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaRxBuffer' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaTxCount' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaTxOut' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaControl' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaRxOut' already defined in module 'cpm22bios_asm'
Error at module 'cpm22_asm': symbol 'aciaTxIn' already defined in module 'cpm22bios_asm'
```

Any input or thoughts on how to resolve this would be welcomed.

suborb, 2021-12-28:
I guess something is pulling in the module that defines those labels?

Adding -Cl-v might help to track it down.

feilipu, 2021-12-28:
Attached the two build logs
[acia listing](https://github.com/z88dk/z88dk/files/7784777/acia.txt)
[acia85 listing](https://github.com/z88dk/z88dk/files/7784778/acia85.txt)

They look very similar, but the acia85 version doesn't complete because of the symbol errors noted above.
The order of library processing seems similar. 
IIRC, the last file found on the command line takes precedence for symbols.

The symbols found in cpm22bios.asm as part of the project should take precedence over library symbols, but they're not for some reason with this hybrid/classic build. Even though they do for newlib builds.

Not sure where to start digging here.

suborb, 2021-12-28:
My hunch at the moment is the forced inclusion by the classic style CRT of a function to satisfy fputc_cons() which may be dragging in the acia labels?

feilipu, 2021-12-28:
If newlib doesn't need to resolve the labels for stdio until "later" in the link process, that could be it. Later in the process, the project provided `PUBLIC` symbols would be available and have precedence.

But I'm just guessing.

feilipu, 2021-12-29:
> My hunch at the moment is the forced inclusion by the classic style CRT of a function to satisfy fputc_cons() which may be dragging in the acia labels.

Totally correct.

commit 07b119aa (2021-12-29): classic - rc2014-8085 closes #1916
closed 2021-12-29 by feilipu