z88dk/z88dk #1918: Unable to compile a simple -subtype=dot program for the zx spectrum

markdryan, 2021-12-29:
I did a clean build of z88dk, commit (1880ff8bb),  yesterday on Ubuntu 20.04.  I built with SDCC, e.g.,

```
cd z88dk
export BUILD_SDCC=1
export BUILD_SDCC_HTTP=1 
chmod 777 build.sh
./build.sh
```

Everything built fine but I'm unable to compile any dot files for the zxspectrum.  Now this is the first time I've tried to do this with z88dk, so I'm probably doing something wrong but I can't figure out what.

Here's my simple program

```
#include <stdio.h>

int main(int argc, char* argv[])
{
	printf("Hello\n");
	return 0;
}
```

which I build with

```
zcc +zx -v  -Cs "--disable-warning 85" -startup=30 -clib=sdcc_iy -SO3 --opt-code-size  hello.c -o hello -m -subtype=dot -create-app
```

And here are the errors I get

```
z88dk-z80asm -I"/opt/z88dk/lib/config/../..//lib/target/zx/def" -D__ESXDOS_DOT_COMMAND -D__SDCC_IY -I"/opt/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/zx" -I"/tmp/tmpzccXXAY4x7K" -D__SDCC  -s  -mz80 -I"/opt/z88dk/lib/config/../..//lib" "/tmp/tmpXXWTKvBH.asm"
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_cmdline_esxdos_argv.inc' line 34: symbol '__SYS_IY' not defined
Errors in source file /opt/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/zx/zx_crt.asm.m4:
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_cmdline_esxdos_argv.inc' line 34: symbol '__SYS_IY' not defined
                   ^ ----          ld iy,__SYS_IY
```

This command works with an earlier version of z88dk from about 8 months ago, although the binaries themselves don't run on the spectrum as the entry address was incorrectly set (https://github.com/z88dk/z88dk/issues/1775).  It was this issue that inspired me to update my version of Z88DK.

If I build for the next instead of the spectrum, i.e., replace the +zx with +zxn, everything builds fine, although I don't have a Next to try the resulting binary.

markdryan, 2021-12-29:
@suborb Thanks for the quick fix.  I should have mentioned in my original bug report that I stumbled across a related issue with subtype=dotx.  When I use this subtype I get an additonal error that complains about __SYSVAR_RAMTOP'

`zcc +zx -v  -Cs "--disable-warning 85" -startup=30 -clib=sdcc_iy -SO3 --opt-code-size  hello.c -o hello -m -subtype=dotx -create-app`

```
z88dk-z80asm -I"/opt/z88dk/lib/config/../..//lib/target/zx/def" -D__ESXDOS_DOT_COMMAND -D__SDCC_IY -I"/opt/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/zx" -I"/tmp/tmpzccXXO5fcvP" -D__SDCC  -s  -mz80 -I"/opt/z88dk/lib/config/../..//lib" "/tmp/tmpXXc1tk8L.asm"
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_load_esxdos_dotx.inc' line 12: symbol '__SYSVAR_RAMTOP' not defined
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_cmdline_esxdos_argv.inc' line 34: symbol '__SYS_IY' not defined
Errors in source file /opt/z88dk/lib/config/../..//libsrc/_DEVELOPMENT/target/zx/zx_crt.asm.m4:
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_load_esxdos_dotx.inc' line 12: symbol '__SYSVAR_RAMTOP' not defined
                   ^ ----    ld hl,(__SYSVAR_RAMTOP)
Error at file '/opt/z88dk/lib/../libsrc/_DEVELOPMENT/target/zx/crt_cmdline_esxdos_argv.inc' line 34: symbol '__SYS_IY' not defined
                   ^ ----          ld iy,__SYS_IY

```

Are you also seeing this?

suborb, 2021-12-29:
Ah yes, I see that. It does suggest there's a whole file missing from +zx that's the equivalent of: https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/zxn/config/config_zxn_sysvar.m4

So I guess that's the answer how to fix things properly...

markdryan, 2021-12-29:
Thanks.  I pulled the latest changes and did a clean rebuild and the problem seems to be fixed.  I can now build both dot and dotx versions of my hello world program without error.  The dot file ran fine on the spectrum but I wasn't able to get the dotx binary to run.  When I issue the CLEAR 32767 statement and then run my dot command, I get an INVALID PATH error.  I've never knowingly run a .dotx command before so this may well be user error.

suborb, 2021-12-29:
I'm similarly clueless about running dotx files - I did think they were just for +zxn nextos rather than the original +zx esxdos.

However, it looks like the original issue is cured and you're up and running with a .dot file so I'm going to close this issue.

closed 2021-12-29 by suborb

aralbrec, 2022-02-08:
dotx programs are a way to build dot commands that are > 8K.  On a standard zx this means part of the dot command is loaded into main ram above ramtop.

The output of a dotx compile is two parts.  The main dot binary should be placed in /bin.  The second part of the dot should be placed in /dot/extra or /dot/extras (can't recall which).  The idea is to keep the /bin directory uncluttered with auxiliary files so that searches for dot commands are not slowed by having those auxiliary files polluting /bin.

The INVALID PATH error is indicating the dot command can't find the second part to load in.