z88dk/z88dk #1920: (z80asm) Is #define with conditionals supported?

suborb, 2021-12-29:
I think I must be doing something wrong:

```
#define ONEBITOUT \
IF SOUND_INOUT = 1 \
    ld      c,a \ \
    and     sndbit_mask \ \
    jr      nz,ASMPC+6 \ \
    in      a,(sndbit_port) \ \
    jr      ASMPC+4 \ \
    out     (sndbit_port),a \ \
    ld      a,c \ \
ELIF SOUND_IFF = 1 \
    ld      c,a \ \
    and     sndbit_mask \ \
    jp      nz,ASMPC+4 \ \
    di \ \
    jp      ASMPC+4 \ \ 
    ei \ \
    ld      a,c \ \
ELIF sndbit_port >= 256 \
    exx \ \
    out     (c),a \ \
    exx \ \
ELIF sndbit_port < 0 \
    ld      (-sndbit_port),a \ \
ELSE \
    out     (sndbit_port),a \ \
ENDIF
```

I've tried various permutations of backslashes and I'm currently at unbalanced control structure and a syntax error at the point where I try to use it.

pauloscustodio, 2021-12-29:
In principle what you did should work after fixing the backslashes - two are needed on each line, one to cause a line break in the macro expansion and another to slurp in the next line while parsing the macro.

It does not work, must be a bug.

Anyway a better and cleaner way is to use the MACRO statement:

```
		MACRO ONEBITOUT
		LOCAL l1, l2, l3, l4
		IF SOUND_INOUT = 1 			
			ld      c,a 			
			and     sndbit_mask 	
			jr      nz,l1
			in      a,(sndbit_port) 
			jr      l2
l1:			out     (sndbit_port),a 
l2:			ld      a,c 			
		ELIF SOUND_IFF = 1 			
			ld      c,a 			
			and     sndbit_mask 	
			jp      nz,l3
			di 						
l3:			jp      l4		 
			ei 						
l4:			ld      a,c 			
		ELIF sndbit_port >= 256 	
			exx 					
			out     (c),a 			
			exx 					
		ELIF sndbit_port < 0 		
			ld      (-sndbit_port),a
		ELSE 						
			out     (sndbit_port),a 
		ENDIF
		ENDM
		
		ONEBITOUT
```

You can assemble with -v -l to show the macro expansion in the list file.

pauloscustodio, 2021-12-29:
Please keep this open for the #define bug.

suborb, 2021-12-29:
Thanks @pauloscustodio - I was playing around with the backslashes to try and get something to work.

The macro does work which will make my life easier for the next odd hardware that comes my way!

commit 285981a6 (2021-12-30): Fix #1920: IF not parsed inside #define macro
Macro call check was only checking for MACRO macros and not #define macros.
commit 73554631 (2021-12-30): Merge pull request #1921 from z88dk/issue_1920

Fix #1920: IF not parsed inside #define macro
closed 2021-12-30 by pauloscustodio