z88dk/z88dk #1925: zcc: output of float constants with wrong size

pauloscustodio, 2021-12-31:
Not sure if this is zcc...

With the command line options `-math-z88` and `-fp-mode=mbf40` zcc should output 5-byte floats but I am getting 6-bytes.

See #305

suborb, 2021-12-31:
I think this was a choice when the non-genmath libraries were added. Basically to keep the size of a float at 6 bytes regardless of the actual size. An even size on the stack makes for easier handling as an added bonus. 

It was only with mbf32/math32 that it made sense to break the relationship. 

I have a feeling that that first byte is just skipped over. 

I think Mathzx/cpc should behave in the same way?

suborb, 2022-01-01:
After a bit more digging we've got:

- CPCmath: 4 bytes mantissa, +128 bias exponent
- ZXmath: 4 bytes mantissa, +128 bias exponent
- Genmath: 5 bytes mantissa, +128 bias exponent
- math48: 5 bytes mantisa, +128 bias exponent
- z88math: 4 bytes mantissa +127 bias exponent

All encode the sign bit as the first bit of the mantissa.

All the libraries pick up the floating value from `fa+1`/address+1 so for zxmath/cpcmath we can just use the 48 bit values with a truncated mantissa.

For z88math the same principle applies, just with a different exponent bias.

Moving onto mbf40, this looks like it uses offset 0-4 in memory, so if there were any libraries we'd pick up from `fa+0`. lIt's written out padded to 6 bytes again though, but byte offset 5 is discarded.

zx70, 2022-03-29:
@suborb,   are you aware of this ?
I'm pasting the link here, I prefer not to open another issue for a probably useless comment :)

https://github.com/cheveron/sebasic4/blob/main/basic/modules/12_calculator.asm