z88dk/z88dk #1931: Memory wrangling

MegaMraz, 2022-01-02:
Probably not an issue, just dumb beginner's question.

I am learning ZX Spectrum development and only reason that I came to C instead of Assembler is registry/mnemonics limitations, too harsh for my tiny head.

I am ready to wrangle memory/interupts/ports, I do not want any functions like "draw_sprite()", hell knows what's inside. I do not want to link any libs at all.

I want to have a set of bytes, just like sjasm's "label: DB 23, 43, 100, 0...." and then memcpy(label, addr), while having all C advantages like unlimited variables, functions and so on.

Can someone give an example of that?

feilipu, 2022-01-02:
> Probably not an issue, just dumb beginner's question.

There are no dumb questions here.

> I do not want any functions like `draw_sprite()`, hell knows what's inside. I do not want to link any libs at all.

If that's the goal, then you're probably not going to find much value in z88dk beyond what sjasm can do for you.

Arguably, the main reason that z88dk exists is to provide a comprehensive and optimised set of libraries to make writing programs in C easier.

If you're going to use C, then you want to use the z88dk library functions to the maximum extent possible. Someone has written them in assembly to be the best they can be. And everyone is open to improving library functions if you have an idea to make them better / faster.

>  `memcpy(label, addr)`

Taking `memcpy()` as an example, depending on what you need it can be [small ](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_memcpy.asm#L100)(as a single `ldir`) or [fast ](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/string/z80/asm_memcpy.asm#L33)like [unrolled `ldi`](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/z80/l_ldi_loop.asm).

IMHO, the right way to write programs for ZX Spectrum is to start by writing what you need in C using library functions for everything you can. Then once things are working correctly you can optimise for speed in assembly language if you find specific things aren't running fast enough.

feilipu, 2022-01-07:
Do re-open if you've some more questions, please.

closed 2022-01-07 by feilipu