z88dk/z88dk #1932: z88dk-ticks - 8085 emulation hangs

feilipu, 2022-01-03:
I've been doing some 8085 [benchmarking](https://github.com/z88dk/z88dk/commit/ab8be5dc79031e6ed4665f488c60c5d25b4e8077), and I'm finding that `z88dk-ticks` is not working when configured with `-m8085`.
Sometimes, it just hangs burning cycles.

Specifically with `pi.c`, compiled using and tested using [these lines](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/pi/z88dk-classic/readme.txt#L30).
```sh
$ zcc +test -m8085 -vn -O2 -DSTATIC -DTIMER -D__Z88DK pi.c -o pi.bin -lndos -m
$ z88dk-ticks -m8085 pi.bin -start 0144 -end 0278 -counter 99999999999
```

The ticks command hangs.

Confirmed [with hardware](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/pi/z88dk-classic/readme.txt#L13) that the benchmark is actually working correctly though, which is good news. :+1:

suborb, 2022-01-03:
Shouldn't the compile line be `-clib=8085` to switch in the libraries? 

```
[Add #pragma printf "%4d" to source code]
dom@ermintrude pi % zcc +test -clib=8085  -vn -O2 -DSTATIC -DTIMER -D__Z88DK -DPRINTF pi.c -o pi.bin -lndos -m
dom@ermintrude pi % z88dk-ticks -m8085 -w 30  pi.bin
3141592653589793238462643383279528841971693993751058209
```

suborb, 2022-01-03:
It's holding up surprisingly well vs the z80.

The size increases are a bit odd - I wonder if I've added too much stuff to the +test crt and it's bloating that way.

feilipu, 2022-01-03:
> `-clib=8085` to switch in the libraries?

Ah. That's probably it. Thought I had interpreted the help info that the `-m8085` switch was sufficient. Do I need both switch and clib?

Yes. Within 1 percent of the z80 scores. I have a much greater appreciation of how well Intel did with it, now. Starting to think z80 design was rushed and underdone in comparison.

But MBF32 is giving me a heartburn. Not at all sure I can beat it. sccz80+mbf32 hold the Whetstone record currently.

suborb, 2022-01-03:
> Ah. That's probably it. Thought I had interpreted the help info that the -m8085 switch was sufficient. Do I need both switch and clib?

No, the clib adds the switch. 

> I have a much greater appreciation of how well Intel did with it, now.

I hope I've got all the timings right. It's amazing the difference two instructions make: stack relative offset + 16 bit load. Adding similar to the z80n would have an made for an interesting performance characteristic with the hardware multiply.

feilipu, 2022-01-03:
I'm surprised the Pi benchmark is so bad in comparison to all the other benchmarks though. It stands out as an oddity.
In real life at 7.3728MHz it takes just over 20 minutes, so the `z88dk-ticks` number is right.

feilipu, 2022-01-03:
> It's amazing the difference two instructions make: stack relative offset + 16 bit load. Adding similar to the z80n would have an made for an interesting performance characteristic with the hardware multiply.

Perhaps the z80n designers didn't understand the value of these additional instructions?

They are pretty well hidden and unless you have used them to generate a purpose built stack relative set of functions it is hard to actually comprehend just how useful they truly are. I think I calculated 3 stack variable accesses for "free" in comparison to moving something from stack to a static location before using it.

Anyway, closing this as the benchmarking is done for now. :+1:

closed 2022-01-03 by feilipu