z88dk/z88dk #1938: sccz80 - aarm64 build

feilipu, 2022-01-08:
Just to to see what happens, I built z88dk on a Clockwork DevTerm.
The architecture is aarm64.
```sh
cpi@clockwork:~/Z80/z88dk$ uname -a
Linux clockwork 5.10.60-rockchip64 #trunk SMP PREEMPT Thu Aug 19 06:58:07 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```
Everything builds, which is more than I expected tbh, but there are some failures with the shift tests for sccz80. Not sure if this is anything more than an interesting exercise. But for the record...
```sh
cpi@clockwork:~/Z80/z88dk$ z88dk-sccz80 -v
Small-C/Plus - z80 Crosscompiler   Version: 1-42569ae2-20220108
(C) 1980-2017 Cain, Van Zandt, Hendrix, Yorston, z88dk
Usage: z88dk-sccz80 [flags] [file]
```
```sh
zcc +test -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c lshift.c -o test_lshift.bin -lndos 
 16:24:25  Starting test 
Starting suite Left shift Tests (6 tests)
Running test test_lshift64_const.....failed lshift.c:330 (<<63)
Running test test_lshift64_var.....failed lshift.c:258 (<<63)
6 run, 4 passed, 2 failed

Ticks: 785060
 16:24:25  Test failure 
make[2]: *** [Makefile:36: test_lshift.bin] Error 1
make[2]: Leaving directory '/home/cpi/Z80/z88dk/test/suites/sccz80'
make[1]: *** [Makefile:24: sccz80] Error 2
make[1]: Leaving directory '/home/cpi/Z80/z88dk/test/suites'
make: *** [Makefile:24: suites] Error 2
make: Leaving directory '/home/cpi/Z80/z88dk/test'
```
```sh
zcc +test -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c rshift.c -o test_rshift.bin -lndos 
 16:25:40  Starting test 
Starting suite Right shift Tests (9 tests)
Running test test_rshift64_const_unsigned.....failed rshift.c:462 (>>0)
Running test test_rshift32_const.....failed rshift.c:170 (>>0)
Running test test_rshift32_var.....failed rshift.c:55 (>>0)
9 run, 6 passed, 3 failed

Ticks: 852804
 16:25:40  Test failure 
make[2]: *** [Makefile:36: test_rshift.bin] Error 1
make[2]: Leaving directory '/home/cpi/Z80/z88dk/test/suites/sccz80'
make[1]: *** [Makefile:24: sccz80] Error 2
make[1]: Leaving directory '/home/cpi/Z80/z88dk/test/suites'
make: *** [Makefile:24: suites] Error 2
make: Leaving directory '/home/cpi/Z80/z88dk/test'
```

suborb, 2022-01-08:
I have a hunch this because sizeof(long double) == 8

I did think I put a warning in about this but it might be only under certain conditions.

feilipu, 2022-01-08:
I' pushed through the tests and the left shift and right shift tests are the only suites that fail (consistently on all intel and zilog), so sccz80 seems pretty damn close to working on aarm64.

Anyway, just mark as "won't fix" and close, if it is not intended to work. 👍

suborb, 2022-01-08:
I’ll take a look. I think arm64 is going to become fairly common architecture so I’d like it to work. 

Hopefully the issue is probably due to some type conversion of the folded constant - there’s been a few instances of this in the past so should be easy to track down.

suborb, 2022-01-08:
So, now I'm looking at the offending tests, it looks like we're invoking the spectre of _undefined behaviour_

```
0x80000000L >> 0 == 0x80000000L (amd64)
0x80000000UL >> 0 == 0x80000000UL (amd64)
0x80000000L >> 0 == 0x7fffffffL (arm64)
0x80000000UL >> 0 == 0x80000000UL (arm64)
```

> 5.8.3

> The value of E1 >> E2 is E1 right-shifted E2 bit positions. If E1 has an unsigned type or if E1 has a signed type and a non-negative value, the value of the result is the integral part of the quotient of E1/2E2. If E1 has a signed type and a negative value, the resulting value is implementation-defined.

feilipu, 2022-01-08:
> it looks like we're invoking the spectre of _undefined behaviour_.

Well, that sucks.

I wonder what Apple do with their M1 implementation? Or is that not even counted as arm64?

suborb, 2022-01-08:
I'm not sure, I don't have access to one unfortunately, and I'm not sure if the behaviour would differ between Linux and macOS on the same hardware.

If you fancy playing around it's primary.c:calc(), I think it's worthwhile adding a diagnostic simply so that I don't have to investigate again, but it's getting the right condition. I get an odd feeling that the sign is flipped in const.c which makes life even more fun.

Interestingly, compiling the same snippet natively behaves "as we would expect":

```
#include <stdio.h>

int main() {
     long val = 0x80000000L;
     int  v = 0;

     printf( "%d %lx %lx", val >> v == 0x80000000UL >> 0, val >> v , 0x80000000L >> 0);
}
```

Though I've not tried a `long long`

dbolli, 2022-01-08:
I compiled the code above on my M1 iMac and I get "1 80000000 7fffffff" printed on screen from running the .tap file created by

zcc +zx -v -O3 -startup=5 -clib=new -m -zorg=32768 --list arm64-long-test.c -o arm64-long-test_CODE.bin
appmake +zx -b arm64-long-test_CODE.bin -o arm64-long-test_CODE.tap --org 32768 --blockname "ARMTEST"

from 7 Jan nightly build

zcc version:
zcc - Frontend for the z88dk Cross-C Compiler - v19112-2f529d2df-20220107

z80asm version:
Z80 Module Assembler 19112-2f529d2df-20220107

suborb, 2022-01-08:
> I compiled the code above on my M1 iMac and I get "1 80000000 7fffffff" printed on screen from running the .tap file created by

Thanks @dbolli That 1 at the start is very unexpected...

```
MacOS/amd64: 

% zcc +test file.c
% z88dk-ticks a.bin
1 80000000 80000000
Ticks: 133220

Linux/arm64:

% zcc +test file.c
% z88dk-ticks a.bin
0 80000000 7fffffff
```

dbolli, 2022-01-08:
I think the 1 at the start is a "true" from "val >> v == 0x80000000UL >> 0"

suborb, 2022-01-08:
It is, but they shouldn't be equal since the next two values aren't the same!

feilipu, 2022-01-09:
Just to add the Rockchip64 gcc 10.3.0 native result.

```sh
cpi@clockwork:~/Z80/src/shift$ gcc shift.c -o shift
cpi@clockwork:~/Z80/src/shift$ ./shift 
1 80000000 80000000
```
And. Oops, the `z88dk-ticks` result.

```sh
cpi@clockwork:~/Z80/src/shift$ zcc +test shift.c -o shift
cpi@clockwork:~/Z80/src/shift$ z88dk-ticks shift
Segmentation fault
```
I'll see if I can get some more information about the segfault.

suborb, 2022-01-09:
> $ gdb z88dk-ticks core
> GNU gdb (Ubuntu 10.1-2ubuntu2) 10.1.90.20210411-git
> ...
> Reading symbols from z88dk-ticks...
> [New LWP 3985]
> Core was generated by `z88dk-ticks shift'.
> Program terminated with signal SIGSEGV, Segmentation fault.
> #0  0x0000ffffb7663aac in strcasecmp () from /lib/aarch64-linux-gnu/libc.so.6
> ```

I think there's some horrible magic about the name of the binary...probably searching badly for a file extension.

suborb, 2022-01-10:
I'm almost there, just very delicate casting needed about the place, this fixes the 32 bit shifts for example:

```
-        else if (left_kind == KIND_LONG) return ((int32_t)left) >> (int)right;
+        else if (left_kind == KIND_LONG) return ((int32_t)(int64_t)left) >> (int)right;
```

And also doesn't break any other tests. I've still to figure out (uint64_t)>>0 and the <<63 failure.

feilipu, 2022-01-10:
If I'm reading correctly, both x86 (amd64) and arm64 will truncate the shift value to 5 or 6 bits. So would it be OK to cast the shift down to `uint8_t`?

And/or perhaps put in a compiler warning too?

suborb, 2022-01-10:
I think there already is a warning for >= sizeof(type)?

suborb, 2022-01-10:
@feilipu - fancy trying out the `arm64_fixups`  branch - would be great to get an independent confirmation before merge.

feilipu, 2022-01-11:
The 64-bit tests still seem to fail.
```

feilipu, 2022-01-11:
The 64-bit tests still seem to fail
```
zcc +test -vn  -I../../framework -DNO_LOG_RUNNING -DNO_LOG_PASSED ../../framework/test.c lshift.c -o test_lshift.bin -lndos 
 11:17:56  Starting test 
Starting suite Left shift Tests (6 tests)
Running test test_lshift64_const.....failed lshift.c:330 (<<63)
Running test test_lshift64_var.....failed lshift.c:258 (<<63)
6 run, 4 passed, 2 failed

Ticks: 785060
 11:17:57  Test failure 
make[2]: *** [Makefile:36: test_lshift.bin] Error 1
make[2]: Leaving directory '/home/cpi/Z80/z88dk/test/suites/sccz80'
make[1]: *** [Makefile:24: sccz80] Error 2
make[1]: Leaving directory '/home/cpi/Z80/z88dk/test/suites'
make: *** [Makefile:24: suites] Error 2
make: Leaving directory '/home/cpi/Z80/z88dk/test'
```

suborb, 2022-01-11:
> The 64-bit tests still seem to fail

What? They pass on my rpi4 -  they were failing with the same errors. Please tell me you forgot to rebuild sccz80!

feilipu, 2022-01-11:
> Please tell me you forgot to rebuild sccz80!

Sorry. Checked it thrice, and it is definitely not working.

All I can surmise is that the Rockchip64 in the [DevTerm A06](https://www.clockworkpi.com/product-page/devterm-kit-a06-series) is not implemented the same as the Broadcom device in the RPi4? But, that's not adding anything. :shrug:  The difference in the result of the test program above would suggest this too.

feilipu, 2022-01-11:
With the exception of the new issue noted above.
All the tests pass. 👍

suborb, 2022-01-11:
Wow, that's a relief to say the least! Thank you. I was asking around to see if I could scrounge a RockChip64 from someone to investigate further. The best I was offered was an OrangePi :)

I do love the fact we've got so many tests, but at the same time...

suborb, 2022-01-11:
Merged as #1946 - will check the snap builds to make sure they're still good

closed 2022-01-11 by suborb

feilipu, 2022-01-12:
Confirmed working on DevTerm A06 - rockchip64 hardware.