z88dk/z88dk #1951: intrinsic l_ret

feilipu, 2022-01-23:
Just noted that the 8085 classic library doesn't provide `l_ret` with `pop hl` instructions prior. So I added one.
This is needed by `asm_strset` an used with `jp Z,l_ret - 1`.

Following up for z80 it seems that two different kinds of `l_ret` are provided. One in the page 0 code __without__ the prior `pop hl` instructions, and one in the library with these pop instructions.

Grepping through the libraries, there are many locations where the `pop hl` instructions are expected, so the page 0 implementations are bad. Don't know why there hasn't been an issue from this?

```
$ rgrep -I l_ret | grep "l_ret -"
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_agile_rcs.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_agile_rcs_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_mega.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_mega.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_mega_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_mega_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_smart_rcs.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_smart_rcs_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_standard.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_standard_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_turbo.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_dzx7_turbo_back.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_sms_dzx7_standard_vram.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/compress/zx7/z80/asm_sms_dzx7_standard_vram_unsafe.asm:        jp      c, l_ret - 1
libsrc/_DEVELOPMENT/drivers/tty/state/asm_tty_state_0.asm:   jp c, l_ret - 1             ; if printable char
libsrc/_DEVELOPMENT/stdio/z80/asm_fread_unlocked.asm:   jp z, l_ret - 1             ; if all records read, return success, hl = nmemb
libsrc/_DEVELOPMENT/string/z80/asm_strset.asm:   jp Z,l_ret - 1
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_768.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_768.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_769.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_769.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_772.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_772.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_773.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_773.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_776.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_776.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_777.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_777.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_784.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_784.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_788.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_788.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_792.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_792.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_798.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_798.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_799.asm.m4:   jp c, l_ret - 3             ; if read error
libsrc/_DEVELOPMENT/target/zxn/startup/zxn_crt_799.asm.m4:   jp nz, l_ret - 3            ; if full amount not read
libsrc/_DEVELOPMENT/env/esxdos/z80/asm_env_getenv.asm:   jp z, l_ret - 2             ; if valsz == 0 return length of value string
libsrc/_DEVELOPMENT/env/esxdos/z80/asm_env_tmpnam.asm:   jp c, l_ret - 2             ; if open failed exit with hl = template
libsrc/compress/zx7/asm_dzx7_mega.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_mega.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_mega_back.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_mega_back.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_standard.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_standard_back.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_turbo.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_turbo_back.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_agile_rcs.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_agile_rcs_back.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_smart_rcs.asm:        jp      c, l_ret - 1
libsrc/compress/zx7/asm_dzx7_smart_rcs_back.asm:        jp      c, l_ret - 1
```

I think the page zero `l_ret` should not be `PUBLIC` for 8085, z80, z180, and zxn.

I'll [delete it from the 8085 page 0](https://github.com/z88dk/z88dk/commit/81cafe335939334e196a9121299f9594bce255dc) implementation. But seeking comment before extending this further to z80, etc,...

feilipu, 2022-01-23:
Better to delete `l_ei_ret` from 8085 and add a proper `l_ret` to align it with z80, etc.

closed 2022-01-23 by feilipu

aralbrec, 2022-02-08:
The library exit points like l_ret are the correct versions.  The crts are packing some of the common library code into gaps in page 0 in order to take advantage of some of the spare space there.  If they do this the crts should have the requisite number of pops before the exit point.

The z80 crt in newlib certainly has it so I'm not sure where the other problems may be:

https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/target/crt_page_zero_z80.inc#L181

commit 6605bce0 (2022-01-23): 8085 - l_ei_ret closes #1951