z88dk/z88dk #1955: Comparison operation 'less or equal' fails with zero

kopitoshka, 2022-01-25:
It seems there is a bug in a standard z88dk compiler.  Here is a sample code returning different results with sdcc and z88dk. The app execution should not change the border, but it does with z88dk compiler. sdcc compiler generates an expected result.

The test was performed on the latest night built ( 2022-01-25 03:09 ).

```
int test(int y) {
	int h = 32;
	
	if (y + h <= 0) {
		return 0;
	}

	return 1;
}

int main(void) {

	int y = -32;

	/* this test function should return 0 but it returns 1 */
	if (test(y)) {
		__asm
		    ld a, 5
			out (254), a	
		__endasm;
	}
}
```

I built that app with these commands.

```
zcc +zx -compiler=sdcc main.c -startup=31 -create-app -o test
zcc +zx main.c -startup=31 -create-app -o test2
```

suborb, 2022-01-25:
Thank you, looks like `rica` should have been used rather than `rla`! I'll commit a fix later on today.

closed 2022-01-26 by kopitoshka

suborb, 2022-01-26:
Thank you for reporting. I'm not sure why our tests didn't pick up on this (since we actually have a test for the exact case). I need to investigate further.

ped7g, 2022-01-26:
you need rogue CF=1 ahead of `rla` to make it fail. ie. it can "usually" work.

suborb, 2022-01-26:
Quite, I did try that last night with an inconvenient `scf`, but wasn't totally focussing so probably messed it up.