z88dk/z88dk #1959: HiTech C v8.0 msdos cross complier

artrag, 2022-01-28:
I have a newer version of HiTech C cross compiler: its note.txt says:

     HI-TECH V8 C Compiler
     Bug fixes made since v7.80

DATE: 26th September 2001

Below is a list of some of the reported bugs that have been fixed since 
version 7.80. Not all code generation bugs are listed since they do not
always directly correspond to a particular piece of code being compiled. 

HPDZ:

-	Fixed incorrect memory summary information. (7.80PL1)

Assembler:

-	Fixed optimization with calls using status register. (7.80PL2)

artrag, 2022-01-28:
Would you interested in using this version to run the comparisons ? I can share its installation files and manuals.
I have used both V8 and V7.5 and the former is optimised in many ways.

feilipu, 2022-01-29:
> Would you interested in using this version (HiTech C) to run the comparisons ?

Certainly.

There are a few cases ([noted on the summary page of Benchmarking](https://github.com/z88dk/z88dk/wiki/Benchmarks)) where the Hitech-C Z80 v7.50 compiler fails to produce a useful result. It would be really interesting to hear whether the v7.80 or v8.00 versions handle the example code better.

If you're able, you can try using the recipes provided in the respective directories (eg for [Whetstone](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/whetstone/htcv750)), and confirm if you get something workable.

Do you know the licencing position with regard to HiTech-C? If it could be problematic, it would be best if you ran the benchmarks, and reported the results in a PR. Otherwise, speaking for myself, I'm sure that we'd be very happy to have copies of v7.8 and / or v8.0 to test with.

Just for cross reference, Tony is maintaining a patched version of [HiTech-C v3.09 for CP/M](https://github.com/agn453/HI-TECH-Z80-C).

artrag, 2022-01-29:
I will send you and email

feilipu, 2022-01-29:
Thanks. Got it.

feilipu, 2022-01-29:
Well, with a bit of mucking about with the various emulators, [dosbox](https://www.dosbox.com/) and the [cpm](https://github.com/jhallen/cpm) tool, I've got some results for the Whetstone benchmark on HiTech-C v7.80p2

And, while they're not totally accurate, they're at least right enough to count!
So, now to work out how to count cycles on this...?

```
$:~/Z80/HITECH-C$ ~/Z80/cpm_tool/cpm/cpm whetston

  
N=      0 J=      0 K=      0
X1=  1.0000e+00 X2= -1.0000e+00
X3= -1.0000e+00 X4= -1.0000e+00

N=    120 J=    140 K=    120
X1= -6.8358e-02 X2= -4.6266e-01
X3= -7.2973e-01 X4= -1.1240e+00

N=    140 J=    120 K=    120
X1= -5.5446e-02 X2= -4.4762e-01
X3= -7.1110e-01 X4= -1.1033e+00

N=   3450 J=      1 K=      1
X1=  1.0000e+00 X2= -1.0000e+00
X3= -1.0000e+00 X4= -1.0000e+00

N=   2100 J=      1 K=      2
X1=  6.0000e+00 X2=  6.0000e+00
X3= -7.1110e-01 X4= -1.1033e+00

N=    320 J=      1 K=      2
X1=  4.9040e-01 X2=  4.9040e-01
X3=  4.9039e-01 X4=  4.9039e-01

N=   8990 J=      1 K=      2
X1=  1.0000e+00 X2=  1.0000e+00
X3=  9.9994e-01 X4=  9.9994e-01

N=   6160 J=      1 K=      2
X1=  3.0000e+00 X2=  2.0000e+00
X3=  3.0000e+00 X4= -1.1033e+00

N=      0 J=      2 K=      3
X1=  1.0000e+00 X2= -1.0000e+00
X3= -1.0000e+00 X4= -1.0000e+00

N=    930 J=      2 K=      3
X1=  8.3464e-01 X2=  8.3464e-01
X3=  8.3464e-01 X4=  8.3464e-01

$:~/Z80/HITECH-C$ 
```

suborb, 2022-01-29:
Ticks can run cpm binaries as long as there’s not too much bdos interaction - just have a .com extension to the binary.

feilipu, 2022-01-30:
> Ticks can run cpm binaries as long as there’s not too much bdos interaction - just have a .com extension to the binary.

Just tried `WHETSTON.COM`, and found following BDOS call issue...
```
~/Z80/HITECH-C$ z88dk-ticks WHETSTON.COM
Unsupported BDOS call 11
^C
~/Z80/HITECH-C$
```
Having a go at other benchmarks now.

feilipu, 2022-01-30:
@artrag Are you sure that this is v8.00?
I see v7.80, with Patch Level 2, reported in quite a few places.
No issue if it is, I just want to label the benchmarking correctly.

[image]

artrag, 2022-01-30:
You are right, but the file txt that come with it says V8 and refers bug fixes after 7.80pl2... It is possible that the version of the sole assembler has changed.I think it is correct refer to v8 even if the IDE says 7.8pl2.

feilipu, 2022-01-31:
So, I've run through all of the benchmarks and [committed the results](https://github.com/z88dk/z88dk/commit/b7edec5421b89c576e149b4a2eaf8e42587d97df). Most of them are working very well.
There's still an issue with [fasta](https://github.com/z88dk/z88dk/tree/master/libsrc/_DEVELOPMENT/EXAMPLES/benchmarks/fasta/htcv780), where the results are not totally correct. However the program runs properly.

So that's all good. Thanks for contributing the compiler :+1:
I'll credit in the wiki.

feilipu, 2022-01-31:
Final update to the wiki now done.

The final Hi-Tech C Compiler posts very good results, and since it can manage banked memory for z180 systems, I'm surprised it isn't in wider use today.

I know that the CP/M v3.09 version is in public domain. It would be good to find out if this version is too.
I called the support numbers, and checked the web site today, but nothing.
The Australian company number still exists though.

closed 2022-01-31 by feilipu

suborb, 2022-01-31:
Can you share the assembler code generated - be interesting to see what it's doing.