z88dk/z88dk #1960: ticks - CP/M getch()

feilipu, 2022-01-31:
For benchmarking with z88dk-ticks for CP/M, I've added in some [additional BDOS calls](https://github.com/z88dk/z88dk/commit/919e3841e992df4ac78503a45129b3b97586e13a).

But, I don't understand [why this `getch()`](https://github.com/z88dk/z88dk/blob/master/src/ticks/hook_cpm.c#L61) hangs waiting for a keystroke, and execution only continues when one is received.

Ideally, with no input, execution should just continue.

suborb, 2022-01-31:
There's a kbhit() in hook_console() that's probably the right thing to use.

feilipu, 2022-01-31:
I've tried with below simple, and more crafty solutions using `kbhit()`, but the result is that instead of proceeding following a keystroke,  the execution echos the keystroke and then hangs.

```c
            int val;
            if ( val = kbhit() != 0 )
                a = l = 1;
            else
                a = l = 0;
```
Is my assumption that there's no need to put the character back on the stream correct?
(I believe that is [happening here](https://github.com/z88dk/z88dk/blob/master/src/ticks/hook_console.c#L39)).

Currently doing this.

```c
            int val;
            if ( val = getch() == EOF )
                val = 0;
            else
                ungetc(val, stdin);
            a = l = val;
```

feilipu, 2022-01-31:
Closed by https://github.com/z88dk/z88dk/commit/48b9dd67327965034da33b74d121ed2a5842ebf3

closed 2022-01-31 by feilipu