z88dk/z88dk #1966: cpm - time() not implemented for INTEL

feilipu, 2022-02-10:
Noted building [zmrx](https://github.com/codesmythe/zmtx-zmrx/blob/main/zmrx.c#L63) for CP/M 8085 that the [`time()`](https://github.com/z88dk/z88dk/blob/master/libsrc/target/cpm/time/time.asm#L25) function is not implemented for INTEL.
The implementation can be a stub since CP/M 2.2 doesn't have time related functions.

zx70, 2022-02-11:
The current time() function includes support for MP/M II, it could be extracted from the Z80 version and used to stub it.
https://github.com/z88dk/z88dk/blob/a74550b04f4e824b3eae024e9a2f46fc234edae8/libsrc/target/cpm/time/time.asm#L148

feilipu, 2022-02-17:
Added a stub for `INTEL` and for Z80 systems less than MP/M-II (i.e. CP/M 2.2) to quickly return 0.
- [x] `time()`
- [x]  `clock()`

closed 2022-02-17 by feilipu