z88dk/z88dk #1967: SDCC 4.2.0 release

spth, 2022-02-10:
Upstream SDCC is working on the 4.2.0 release, aiming for a release in Feburary. sdcc trunk is soft-frozen as of today.

Since SDCC releases tend to be relatively stable, it might make sense for z88dk to consider using the 4.2.0 release in the future. If z88dk wants to do so, it might make sense to check current sdcc trunk for any issues that might affect z88dk, and could still be fixed before the release.

Philipp
(SDCC 4.2.0 release manager)

feilipu, 2022-02-11:
Would it be possible to patch the z88dk-zsdcc build to default to the old calling style?

Or perhaps more clearly put, could you suggest the simplest way to patch the default to `__sdcccalling(0)`?

spth, 2022-02-11:
The default can be set via a command line option, e.g. --sdcccall 0.

feilipu, 2022-02-11:
The default needs to be just that, the default.

Everything in z88dk will be broken if the default (i.e. no change to ABI) is no longer the default.
 - all the libraries, tests and examples make files will be broken if that changes.
 - everything application or library anyone has done prior and from now with blended assembly and C will be broken if the default changes.

I think the patches to the `ABI revision` level as noted above are sufficient to correct the default ABI.
Certainly minor testing I've done today would suggest that to be the case.

If there are additional places that we should patch to regain the default ABI revision, then please advise.

spth, 2022-02-11:
Your changes should be sufficient, to make __sdcccall(0) the default again.

However, in the long term, I suggest z88dk migrate to the new sdcc default, as there should be a substantial performance and code size advantage.

spth, 2022-02-11:
Have a look at the bug-2712.c and muldiv.c regression tests in SDCC for example code to generate calls to these.

feilipu, 2022-02-13:
Just checking back on our issues list and #1129, together with the underlying SDCC reports, are still open.

- https://sourceforge.net/p/sdcc/bugs/2152/
- https://sourceforge.net/p/sdcc/bugs/2733/
- https://sourceforge.net/p/sdcc/bugs/2877/

feilipu, 2022-02-15:
>  P.S.: A bit of data on how much the code improves using __sdcccall(1): https://arxiv.org/abs/2112.01397

It is a very useful analysis.

But, I am missing the comparison between the new register calling convention and the existing `__z88dk_fastcall` convention (`dehl` register passing), which would be very relevant to the existing capabilities of sdcc and would show the real world improvement possible either with and/or without optimised library functions.

Excluding the existing register passing mechanism from the comparison with the new register passing mechanism seems to tie "one arm behind the back" of the sdcc existing capability. Or, did I miss that section?

> Stack parameters were always cleaned up by the caller. The original rationale for the choice of the calling convention for Z80 (the oldest version of SDCC we found, SDCC 2.2.0 from 20 years ago already supports the Z80, and the calling convention hasn’t changed since).

Just for interest, I've spent the past months working with the 8085 CPU for the first time. Given the paucity of registers it is really difficult to write efficient reentrant callee functions. Having the caller tidy the stack after dealing with the return values saves a lot of unnecessary [gymnastics](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/7-8085/i32/l_long_div_m.asm#L38).

Also, for writing reentrant functions 8085 it is preferable to have the parameters provided on the stack, otherwise the first thing any function has to do is to clear the (very few) registers to the stack so that there are pointers available. Fragments like following are common.
```asm
    ld a,(bc)
    sbc (hl)
    ld (de),a
```
:thought_balloon: So my thinking is that the original SDCC authors had similar experiences from their 8080 / 8085 past, and brought those thoughts into the early the SDCC design. That's why it has been stack based since the beginning. :thinking:

spth, 2022-02-16:
> Ok @spth, checking back on our issues list against sdcc for the stable v4.2.0 release. I've triaged and closed all that I can at this stage.
> 
> The remaining issues are on the sdcc side for your stable release.
> 
> First #1129, together with the underlying SDCC reports, are still open (3 years).
> 
>     * [ ]  https://sourceforge.net/p/sdcc/bugs/2152/
> 
>     * [ ]  https://sourceforge.net/p/sdcc/bugs/2733/
> 
>     * [ ]  https://sourceforge.net/p/sdcc/bugs/2877/
> 
> 
> And also #898 (for 4 years) with
> 
>     * [ ]  https://sourceforge.net/p/sdcc/bugs/2810/

I don't think we'll be able to fix those old issues for 4.2.0.

> 
> And finally #1914 is an odd regression, that was working with r12069 but now doesn't.
> 
>     * [x]  https://sourceforge.net/p/sdcc/bugs/3322/

We have a fix for this regression, that should go in before we prepare RC1.

Philipp

feilipu, 2022-02-16:
> We have a fix for this #1914 regression, that should go in before we prepare RC1.

Great. I pulled the patch into the PR to test. It will disappear once it's in the Release, before merging.

spth, 2022-02-16:
> Just for interest, I've spent the past months working with the 8085 CPU for the first time. Given the paucity of registers it is really difficult to write efficient reentrant callee functions. Having the caller tidy the stack after dealing with the return values saves a lot of unnecessary [gymnastics](https://github.com/z88dk/z88dk/blob/master/libsrc/_DEVELOPMENT/l/sccz80/7-8085/i32/l_long_div_m.asm#L38).
> 
> Also, for writing reentrant functions for the 8080 / 8085 it is preferable to have the parameters provided on the stack, otherwise the first thing any function has to do is to clear the (very few) registers to the stack so that there are pointers available. Fragments like following are common.
> 
> ```assembly
>     ld a,(bc)
>     sbc (hl)
>     ld (de),a
> ```
> 
> thought_balloon So my (computer archeology) thinking is that the original SDCC authors had similar experiences from their 8080 / 8085 past, and brought those thoughts into the early the SDCC design. And, that's why it has been stack based caller clearing since the beginning. thinking

The paper has some data on SM83, which here would be closer to 8080 than Z80 (after all, SM83 has neither ix nor iy). However, it turned out that even for SM83, current SDCC can use some register parameters well. As long as hl is never used for such a parameter (since we'll need free hl to access memory, especially the stack).

spth, 2022-02-16:
> It is a very useful analysis.
>
> But, I am missing the comparison between the new register calling convention and the existing `__z88dk_fastcall` convention (`dehl` register passing), which would be [...]

The __z88dk_fastcall calling convention is indeed not explicitly mentioned there. AFAIK, outside of z88dk it never found widespread use. And it is limited to one argument only.
However, in the big comparison, calling conventions that for functions with just one parameter are equivalent to __z88dk_fastcall were considered (and also calling conventions that behave like __z88dk_fastcall only for functions with 8 bit arguments or only 16 bits arguments or 32-bit arguments, or just two out of those, calling convention that combine returning like __z88dk_fastcall with other argument passing or the other way around, etc). However, it simply didn't seem feasible to present detailed data on all the thousands of calling conventions used in the experiments.

feilipu, 2022-02-17:
> However, in the big comparison, calling conventions that for functions with just one parameter are equivalent to `__z88dk_fastcall` were considered... However, it simply didn't seem feasible to present detailed data on all the thousands of calling conventions used in the experiments.

No of course presenting data from thousands of experiments would be impractical and irrelevant.

Just it would seem to be useful to present comparison data from the register passing option that sdcc has already implemented, i.e. `__z88dk_fastcall`, and compare that to the new register passing mechanism. That would be somewhat of an "apples to apples" comparison; rather than comparing the existing stack based parameter methods to the new register passing proposal. Obviously register passing will produce a better outcome.

But, I think you did do that analysis that for us previously though in #1827, right? :+1: 
So unless there's a substantial change, then there's no need to repeat that effort.

feilipu, 2022-02-19:
Invalid opcode generation
- [ ] https://sourceforge.net/p/sdcc/bugs/3327/

feilipu, 2022-02-22:
For your interest, I've gone though all of our benchmark examples with the vanilla 4.2.0 r13050 RC1 and have proven them to work (as expected).

There have been some substantial improvements across the board. I'd guess that most of that comes from the new calling convention, and from the new `___mulsint2slong` intrinsic written assembly (particularly for the mandelbrot result).

There are lots of [files to check](https://github.com/z88dk/z88dk/pull/1969/files), so the easiest way to get a overview might be in the [Wiki with a comparison](https://github.com/z88dk/z88dk/wiki/Benchmarks/_compare/18a7e1ee18cf76b757821a745255594a6e7ffaee...c6a521ad520a91f6dd419ecceeb4d5739d029c3a) of the edit I just made.

When I feel up to it I'll go back through and update all the zsdcc benchmarks too. Though I don't expect there to be much movement there.

PR #1969 (2022-03-09): zsdcc - migrate to stable release v4.2.0
Closes #1470
Closes #1827
Closes #1861
Closes #1914
Closes #1967
Closes #1977

Release of upstream [SDCC v4.2.0](https://sourceforge.net/p/sdcc/code/13131/).

__ADDITIONS__

- [x] Need to patch the default calling for all z88dk implemented architectures to `__sdcccall(0)`<br>to ensure no changes are necessary within z88dk libraries.
- [x] Need to provide new intrinsic functions: `___mulsint2slong`,  `___muluint2ulong`.
- [x] Add new intrinsic signed function `___mulsint2slong` to `am9511` math library<br>(however not unsigned, which remains in `l_` libr [... cut]
closed 2022-03-09 by feilipu