z88dk/z88dk #1970: [question] .map file now always generated. Is this a bug? How can I avoid this?

Fabrizio-Caruso, 2022-02-13:
With the latest builds I see that the .map file is now always generated on all targets. Is this wanted or a bug?
Is there an option to avoid this? If not, please, consider adding an option for this or, better, have the map file generated only upon request as before.

feilipu, 2022-02-13:
A `.map` is generated by the `-m` option, or when `-create-app` is used.
The `-create-app` needs the map file to do its magic.

Fabrizio-Caruso, 2022-02-13:
@feilipu  the .map was removed automatically in previous versions when the -m option was not provided even if the -create-app option was used (as in all my targets that are more up to 100 per single game).
I do not see why now the .map should be left when there is no explicit desire from the user.
I can fix it by removing it myself. I will have to change 100 targets on several files...
If there is no reason to change the old behavior, I would prefer not to have to delete the .map file in my scripts.

feilipu, 2022-02-13:
Sorry, don't know if the behaviour was changed purposefully. :man_shrugging:

Fabrizio-Caruso, 2022-02-13:
The problem is that this change affects all targets. So any user will now have to clean this up for any target they were already using. Functionally .map output makes sense for debugging whereas invoking appmake is meant for quite the opposite, i.e. to transform a raw binary into something ready to use.
In general I also wonder why temporary files are not automatically removed when no explicit option is provided.

suborb, 2022-02-13:
I think this behaviour changed a couple of months ago - up until that point z80asm created the map file in /tmp, after that it created the .map alongside the binary.

The removal if `-m` wasn't supplied wasn't added unfortunately.

commit de021b21 (2022-02-13): Fix #1970: Remove .map file if it's not needed
closed 2022-02-13 by suborb