z88dk/z88dk #1971: [vz200] Support for .wav 22khz format for VZEM

Fabrizio-Caruso, 2022-02-15:
Binaries .VZ files built by appmake do not work correctly under VZEM with a bug that is probably caused by VZEM.
.WAV files produced by Z88DK/appmake cannot be used with VZEM, which complains with error about the 22khz, mono 8-bit format.

The bug has already been reported here: when loading my games if I use the keyboard, the keys are echoed on the screen.

How can I build .wav files with Z88DK in VZEM format (22khz, mono 8-bit)?

[image]

suborb, 2022-02-15:
Have you tried the option -Cz-22 ? - it looks like that turns on 22050Hz mode in appmake.

Fabrizio-Caruso, 2022-02-15:
@suborb , no because I did not find it in
https://github.com/z88dk/z88dk/wiki/Platform---VZ200

Is there a general way to know which options are available for each target?

Fabrizio-Caruso, 2022-02-15:
@suborb I have compiled with `-Cz-22 -Cz--audio` and the .wav is not read by VZEM :-(

Moreover VZEM seems to produce random keyboard echo when used with .VZ files produced by Z88DK.

suborb, 2022-02-15:
> Is there a general way to know which options are available for each target?

`z88dk-appmake +[platform]` will show the options available.

Fabrizio-Caruso, 2022-02-15:
@suborb Does `z88dk-appmake +[platform]` provide all the options that are also available when using `zcc` ? or only the once that `appmake` provides as a stand-alone application?

suborb, 2022-02-16:
All the options are available all the time, zcc uses `-c ` `-b` etc but you should be able to pass any option through with -Cz

feilipu, 2022-05-11:
Possibly #2013 related?

zx70, 2022-05-11:
Maybe.   I just corrected the WAV format a little bit.
The VZ200 options already included "-Cz--22" to reduce the rate:

$ z88dk-appmake +vz --help
appmake +vz (vz2cas)

(C) 2010 Stefano Bodrato

Convert the Laser 200 .vz file to .cas, optionally to WAV

Options:

-h   --help            (bool)    Display this help
-b   --binfile         (string)  Linked binary file
-c   --crt0file        (string)  crt0 file used in linking
-o   --output          (string)  Name of output file
     --audio           (bool)    Create also a WAV file
     --fast            (bool)    Create a fast loading WAV
     --22              (bool)    22050hz bitrate option
     --blockname       (string)  Name of the code block

closed 2022-10-20 by zx70