z88dk/z88dk #1972: Nightly build for macOS fails with error message "make[1]: *** [Issue_1623_stdcbench_issues.opt] Error 1"

dbolli, 2022-02-15:
Nightly build "make clean" for macOS 

`OS="darwin"
export OS
ARCH="arm64"
export ARCH

sudo ./build.sh
`

fails with the following error message

`diff -w Issue_1623_stdcbench_issues.opt.tmp2 results/Issue_1623_stdcbench_issues.opt
94c94
< 	call	l_long_mult
---
> 	call	l_long_mult_u
make[1]: *** [Issue_1623_stdcbench_issues.opt] Error 1
make: *** [testsuite] Error 2
`

suborb, 2022-02-15:
I think for weird historic reasons `make clean` doesn't clean out the binaries in bin

Issue #1954 recently introduced a theoretical difference between signed and unsigned multiplication which is what is being shown up here.

In short: remove the binaries in bin/ and try again

dbolli, 2022-02-15:
"sudo make clean" has worked in the past and was working up until a couple of weeks ago (i.e. the last time I downloaded a nightly build for macOS)
 
```
OS="darwin"
export OS
ARCH="arm64"
export ARCH

sudo ./build.sh

```

the build script runs successfully.

closed 2022-02-16 by feilipu

dbolli, 2022-02-17:
I deleted the z88dk/bin/ dir and re-ran "sudo make clean" and it failed with several error messages when trying to compile scan_rules.c. I will raise a new issue for this error.

```
gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/scan_rules.o src/c/scan_rules.c
src/c/scan_rules.c:89:1: error: expected identifier or '('
%%{
^
src/c/scan_rules.c:254:2: error: expected identifier or '('
}%%
 ^
src/c/scan_rules.c:258:45: error: unknown type name 'bool'
static void set_scan_buf( const char *text, bool _at_bol )
                                            ^
src/c/scan_rules.c:260:2: error: implicit declaration of function 'Str_set' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        Str_set( input_buf, text );             
        ^
src/c/scan_rules.c:260:11: error: use of undeclared identifier 'input_buf'
        Str_set( input_buf, text );             
                 ^
src/c/scan_rules.c:261:2: error: use of undeclared identifier 'p'
        p = Str_data(input_buf);
        ^
src/c/scan_rules.c:261:6: error: implicit declaration of function 'Str_data' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        p = Str_data(input_buf);
            ^
src/c/scan_rules.c:261:15: error: use of undeclared identifier 'input_buf'
        p = Str_data(input_buf);
                     ^
src/c/scan_rules.c:264:2: error: use of undeclared identifier 'at_bol'; did you mean '_at_bol'?
        at_bol  = _at_bol;
        ^~~~~~
        _at_bol
src/c/scan_rules.c:258:50: note: '_at_bol' declared here
static void set_scan_buf( const char *text, bool _at_bol )
                                                 ^
src/c/scan_rules.c:265:2: error: use of undeclared identifier 'pe'
        pe              = Str_data(input_buf) + Str_len(input_buf);
        ^
src/c/scan_rules.c:265:30: error: implicit declaration of function 'Str_len' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                                ^
src/c/scan_rules.c:265:30: note: did you mean 'Str_set'?
src/c/scan_rules.c:260:2: note: 'Str_set' declared here
        Str_set( input_buf, text );             
        ^
src/c/scan_rules.c:265:17: error: use of undeclared identifier 'input_buf'
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                   ^
src/c/scan_rules.c:265:38: error: use of undeclared identifier 'input_buf'
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                                        ^
src/c/scan_rules.c:266:2: error: use of undeclared identifier 'eof_'
        eof_    = pe;   /* tokens are not split acros input lines */
        ^
src/c/scan_rules.c:266:9: error: use of undeclared identifier 'pe'
        eof_    = pe;   /* tokens are not split acros input lines */
                  ^
src/c/scan_rules.c:268:2: error: expected expression
        %%write init;
        ^
src/c/scan_rules.c:268:3: error: expected expression
        %%write init;
         ^
src/c/scan_rules.c:268:4: error: use of undeclared identifier 'write'; did you mean 'while'?
        %%write init;
          ^~~~~
          while
src/c/scan_rules.c:268:4: error: expected expression
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [src/c/scan_rules.o] Error 1
make: *** [bin/z88dk-z80asm] Error 2

```

dbolli, 2022-02-17:
As per [pauloscustodio](https://github.com/pauloscustodio) answer to Issue https://github.com/z88dk/z88dk/issues/1974, I have downloaded the latest nightly build for macOS now that ragel is installed, deleted z88dk/bin/ before running "sudo make clean" and it runs successfully.