z88dk/z88dk #1973: Nightly build for macOS "sudo make install" fails with error message "Can't exec "ragel": No such file or directory at parse_ragel.pl line 56."

dbolli, 2022-02-15:
Nightly build "sudo make clean" for macOS fails with the following error message

```
perl parse_ragel.pl src/c/scan_rules.rl
Can't exec "ragel": No such file or directory at parse_ragel.pl line 56.
'ragel -T0 -osrc/c/scan_rules.h2 src/c/scan_rules.rl2' failed: No such file or directory
make[1]: *** [src/c/scan_rules.h] Error 2
make: *** [install] Error 2

```

suborb, 2022-02-16:
You need to have the following installed when building from source: https://github.com/z88dk/z88dk/wiki/installation#building-from-sources

dbolli, 2022-02-17:
I have installed ragel via homebrew and the above error still occurs

```
$ ragel -v
Ragel State Machine Compiler version 6.10 March 2017
Copyright (c) 2001-2009 by Adrian Thurston

```

dbolli, 2022-02-17:
As per @pauloscustodio [pauloscustodio](https://github.com/pauloscustodio) answer to Issue #1974, I have downloaded the latest nightly build for macOS now that ragel is installed, deleted z88dk/bin/ before running "sudo make clean" and it runs successfully. Can I please suggest that when there are new dependencies, they are announced on the https://github.com/z88dk twitter feed? Not everybody who downloads the nightly build regularly checks the installation/Building from sources page on GitHub. Re-tested successfully. Closed.

closed 2022-02-17 by dbolli