z88dk/z88dk #1974: Nightly build for macOS "sudo make clean" fails with error message "src/c/scan_rules.c:89:1: error: expected identifier or '('"

dbolli, 2022-02-17:
Nightly build "sudo make clean" for macOS fails with the following error message

```
gcc -std=gnu11 -MMD -Wall -O3 -Isrc -Isrc/c -It -I../common -I../../ext/optparse -I../../ext/regex -I../../ext/uthash/src  -g -O2 -c -o src/c/scan_rules.o src/c/scan_rules.c
src/c/scan_rules.c:89:1: error: expected identifier or '('
%%{
^
src/c/scan_rules.c:254:2: error: expected identifier or '('
}%%
 ^
src/c/scan_rules.c:258:45: error: unknown type name 'bool'
static void set_scan_buf( const char *text, bool _at_bol )
                                            ^
src/c/scan_rules.c:260:2: error: implicit declaration of function 'Str_set' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        Str_set( input_buf, text );             
        ^
src/c/scan_rules.c:260:11: error: use of undeclared identifier 'input_buf'
        Str_set( input_buf, text );             
                 ^
src/c/scan_rules.c:261:2: error: use of undeclared identifier 'p'
        p = Str_data(input_buf);
        ^
src/c/scan_rules.c:261:6: error: implicit declaration of function 'Str_data' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        p = Str_data(input_buf);
            ^
src/c/scan_rules.c:261:15: error: use of undeclared identifier 'input_buf'
        p = Str_data(input_buf);
                     ^
src/c/scan_rules.c:264:2: error: use of undeclared identifier 'at_bol'; did you mean '_at_bol'?
        at_bol  = _at_bol;
        ^~~~~~
        _at_bol
src/c/scan_rules.c:258:50: note: '_at_bol' declared here
static void set_scan_buf( const char *text, bool _at_bol )
                                                 ^
src/c/scan_rules.c:265:2: error: use of undeclared identifier 'pe'
        pe              = Str_data(input_buf) + Str_len(input_buf);
        ^
src/c/scan_rules.c:265:30: error: implicit declaration of function 'Str_len' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                                ^
src/c/scan_rules.c:265:30: note: did you mean 'Str_set'?
src/c/scan_rules.c:260:2: note: 'Str_set' declared here
        Str_set( input_buf, text );             
        ^
src/c/scan_rules.c:265:17: error: use of undeclared identifier 'input_buf'
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                   ^
src/c/scan_rules.c:265:38: error: use of undeclared identifier 'input_buf'
        pe              = Str_data(input_buf) + Str_len(input_buf);
                                                        ^
src/c/scan_rules.c:266:2: error: use of undeclared identifier 'eof_'
        eof_    = pe;   /* tokens are not split acros input lines */
        ^
src/c/scan_rules.c:266:9: error: use of undeclared identifier 'pe'
        eof_    = pe;   /* tokens are not split acros input lines */
                  ^
src/c/scan_rules.c:268:2: error: expected expression
        %%write init;
        ^
src/c/scan_rules.c:268:3: error: expected expression
        %%write init;
         ^
src/c/scan_rules.c:268:4: error: use of undeclared identifier 'write'; did you mean 'while'?
        %%write init;
          ^~~~~
          while
src/c/scan_rules.c:268:4: error: expected expression
fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
make[1]: *** [src/c/scan_rules.o] Error 1
make: *** [bin/z88dk-z80asm] Error 2

```

pauloscustodio, 2022-02-17:
This error is a consequence of a failed generation of src/c/scan_rules.c (e.g. by the ragel tool not being found).
The solution is to remove src/c/scan_rules.c and call make again.

dbolli, 2022-02-17:
Thanks Paulo, I have downloaded the latest nightly build for macOS now that ragel is installed, deleted z88dk/bin/ before running "sudo make clean" and it runs successfully. Can I please suggest that when there are new dependencies, they are announced on the @z88dk twitter feed? Not everybody who downloads the nightly build regularly checks the installation/Building from sources page on GitHub. Re-tested successfully. Closed.

```
$ z80asm -h
Z80 Macro Assembler 19345-4cc4eb043-20220217
(c) InterLogic 1993-2009, Paulo Custodio 2011-2022

```

closed 2022-02-17 by dbolli