z88dk/z88dk #1975: No support for the Corvette

litwr2, 2022-02-17:
I want to add this support but I don't know how to start.  Any help?  It is about https://en.wikipedia.org/wiki/Corvette_(computer)
I am not sure that this place is proper for my question.  So please help me to find a correct one if I confused something.

suborb, 2022-02-20:
I'm sorry I genuinely got distracted by other things and forget to reply!

As I read it, the Corvette can run in a couple of different ways: via CP/M and via BASIC.

It's going to be easiest to get up and running using CP/M and then re-use what you've written for the BASIC route.

So for CP/M the usual route I take is:

1. Update appmake to create a disc image with the correct format: https://github.com/z88dk/z88dk/blob/master/src/appmake/cpm2.c
2. Update cpm.cfg to reference that format: https://github.com/z88dk/z88dk/blob/master/lib/config/cpm.cfg

At this point, you should now be able to issue:

    zcc +cpm -clib=8080 -subtype=corvette [files] -create-app

And a disc image is created which can be written to physical media and the program on the disc executed on the real hardware.

At this point in time, everything is using the BDOS or BIOS so we're not taking advantage of any of the hardware of the machine. The first thing I do now is to port the VT52 engine to the hardware. I can see there's a text mode supported, so start with that. 

You need really few endpoints to get it working: clear screen, scroll up one row, print character, "peek" character, handle attributes. There's an example for the Mattel Aquarius here: https://github.com/z88dk/z88dk/blob/master/libsrc/target/aquarius/stdio/generic_console.asm 

Depending on the chargen at this point you should be able to get lores graphics up and running (take a look at the mikro80 for example).

After that, it's moving onto the graphical modes and implementing the console and getting hires graphics running (Lviv or vector06c might be good examples).

Sound support is also possible (either PSG or one bit sound) and support for any hardware joysticks.

The final bits to round it off are usually "inkey" keyboard reading which handles multiple keypresses and any serial port hardware.

For the BASIC side you should be able to re-use all the library code you wrote, but you'll also need:

1. Update appmake to support creating tape files: I'm wondering if the MSX file generator could be reused
2. Create a crt0 for the target
3. Create a corvette.cfg file that brings it all together

If you look through the issues there's quite a few "Add target" ones and following the commits should show what's needed. Hopefully the structure of where files go is intuitive.

Finally, if you want to contribute your code, a wiki page is really appreciated: basically a hardware summary, supported features summary and how to load and run a program and links to any emulators or background information on the hardware.

litwr2, 2022-02-22:
Thank your very much for this valuable information.  I have done several first steps and got some results.  The system can generate an executable on a disk image now.  However I have encountered several problems:

1. when I generated Z88DK first time I got an error so I have to fix *src/z88dk/src/z80asm/src/cpp/utils.h*, I had to add

```
#include <iomanip>
```
to it;

2. Besides editing *src/z88dk/lib/config/cpm.cfg* and *src/z88dk/src/appmake/cpm2.c* I had to add some information to *src/z88dk/src/appmake/appmake.h* and *src/z88dk/support/multitarget_build/Makefile*;

3. it seems that the Corvette disks can't work without proper system tracks, the system tests a disk checksum when it has an access to the disk.  The system can use several disk formats and, I suppose, it must read disk parameters from the system tracks.  How to add information to the system tracks to a disk image in Z88DK?

4. how to contribute?  Should I start a new branch?

Thank you.

litwr2, 2022-02-22:
Actually we need to write 20 bytes to the first track that allows the Corvette to detect a format and use this disk.

suborb, 2022-02-22:
Re `<iomanip>` thanks for confirming this - just out of interest what OS/compiler versions are you using?

Boot sectors are handled already: if you look at `struct formats` in cpm2.c you'll see there's bootlen/bootsector fields for specifying the contents of the boot sector. Similarly if some files need to be present on disc the `extra_hook` field deals with that. There's examples of the usage of both of these features.

If you fork the project, create a branch and then raise a PR with the changes.

litwr2, 2022-02-22:
I use **gcc (Debian 8.3.0-6) 8.3.0** under **Linux my2 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux**.
Thank you.  I have added 32 bytes to the boot sector and all things start to work.
Is it correct that I need to fork **z88dk** to my github account?

litwr2, 2022-02-22:
I use **gcc (Debian 8.3.0-6) 8.3.0** under **Linux my2 4.19.0-14-amd64 #1 SMP Debian 4.19.171-2 (2021-01-30) x86_64 GNU/Linux**.
Thank you.  I have added 32 bytes to the boot sector and all things start to work.  So any simple console programs may be successfully compiled for the Corvette.
Is it correct that I need to fork **z88dk** to my github account?

suborb, 2022-02-22:
Yes, it's correct that you fork to your account and then raise a PR back with changes.

litwr2, 2022-02-23:
It is done - [https://github.com/litwr2/z88dk](url)
Now I am going to realize the vt52 emulation.

feilipu, 2022-02-23:
For convenience, once you've made a fork, remember to create a new branch to work on a PR within.
Then your fork "master" branch can then be kept up to date with the z88dk/master.
And then it remains easy to keep your PR branch up to date. For the eventual PR it is also then easy to propose.

litwr2, 2022-02-23:
Thank you.  I have made branch corvette and put all my contributions there.  But what about Wikipedia?

feilipu, 2022-02-23:
Probably best to put any modifications to the wiki here, and someone can add them where appropriate.
You can use markdown within these comments, like in the wiki too.

litwr2, 2022-06-01:
It is typical to use the KDI filename extension for the Corvette disk images.  Is it possible to use it as default for platform `corvette' instead of DSK?
Is there a list to a known program sources that may be compiled with ZCC?  Now I am seeking programs with console io only (CP/M), this can be used for testing.  For instance, where to find Adventure-game sources which can be compiled with ZCC?
Thanks.

suborb, 2022-06-05:
What format is kdi? If it's not just a rename of an existing format then a new writer would have to be written.

In terms of adventures, there's adv_a.c in the repo already, and @zx70 has a port of ScottFree somewhere.

litwr2, 2022-06-06:
KDI is just another name, it is used as the default extension by some (most?) Corvette software (emulators, disk image utilities, etc). So is it just a possibility to use the KDI extension?

litwr2, 2022-06-08:
> look at z88dk/lib/config/cpm.cfg there 's a bit of disk configurations, some with customised file extensions
Thank you. It works!

I still need more help.  I am trying to compile game "adventure" for the Corvette but I can't still overcome some issues. :( The next command works
    zcc +cpm -clib=8080 -lm  adv_a.c
it produces a BIN-file, I change the extension to COM, put it on a disk image, and it works.  There is only one problem, the Corvette CP/M uses code #31 to clear screen, not #12.  How can I resolve this?  Are there other control codes (except CR, LF) that are handled by basic CPM/M-console?
The next command
    zcc +cpm -subtype corvette -lm -o adventure -create-app adv_a.c
produces a COM-file and KDI disk image.  But this COM-file crashes. :( What should I fix?
I have one more question.  Is it possible to use several disk formats?  I have implemented the data format that is not bootable, it uses only one system track (actually, it needs to fill only 32 first bytes).  Bootable formats use 2 or 3 system tracks.
I have made an initial Wiki-page at https://github.com/z88dk/z88dk/wiki/Platform---Corvette.
I can also report about a small issue.  Rebuild doesn't work properly if I didn't explicitly delete "z88dk-appmake" before.

litwr2, 2022-06-09:
I have been able to handle the issue with the clear screen.  I had to edit libsrc/target/cpm/stdio/fputc_cons.asm, I made the next changes
```
    cp	12	; FF (CLS) ?
    jp nz,next   ;corvette only
    ld e,31       ;corvette only
    ld      c,2    ;corvette only
    jp	__bdos ;corvette only
    ;jr	z,docls  ;not for the corvette
.next
```
But it seems this is rather a hack than a real solution.  Is there a better way to get the same result?

zx70, 2022-06-09:
Your're at a good stage already :)
There are several ways to deal with the display peculiarities, z88dk includes even advanced terminal emulation elements you could reuse. 
I'd suggest you to leave it behind and focus on cleaning up the configuration and the disk formats and open a first pull request.

litwr2, 2022-06-10:
Thank you very much!  I have just created my first pull request.
I still need some help about:
1) Is it possible to use several disk formats? I have implemented the data format that is not bootable, it uses only one system track (actually, it needs to fill only 32 first bytes). Bootable formats use 2 or 3 system tracks;
2) How to implement proper screen clear procedure?  The Corvette uses code 31 instead of 12 to clear screen.  The edit of libsrc/target/cpm/stdio/fputc_cons.asm helps but it is a hack...

zx70, 2022-06-10:
You can provide different disk formats.
I think I did it on the Amstrad PCW,  see the 2 lines here..
https://github.com/z88dk/z88dk/blob/b30d797c1c1d4ccf08447f74e481a43ff9a19917/lib/config/cpm.cfg#L33
And the 2 definition blocks starting here...
https://github.com/z88dk/z88dk/blob/b30d797c1c1d4ccf08447f74e481a43ff9a19917/src/appmake/cpm2.c#L127

They also include special data in the boot sector, which is defined here:
https://github.com/z88dk/z88dk/blob/b30d797c1c1d4ccf08447f74e481a43ff9a19917/src/appmake/cpm2.c#L569

We have a sort of a standard behavior of the basic console commands in z88dk, where code $12 is often used to clear the screen.  This brought us to often keep it as the valid code and eventually remap it to the desired subroutine / code remapping fn.
What the CP/M function you are modifying does, is to look for the character 12 and jump to 'docls' where the ANSI CLS sequence (27, '[', 'J') is issued.
You could, in example, create a target specific library which includes a special fputc_cons version.  In the linking phase this could be used to override the original one.

By the way z88dk permits you to force the console output to a different function leaving the original one untouched.
A similar mechanism permitted us to provide both support for the VT/ANSI and ADM 3, the different CLS sequence is here:
https://github.com/z88dk/z88dk/blob/master/libsrc/target/cpm/stdio/generic_console.asm

If your computer does not support the ADM 3 or the ANSI VT protocols, you could be interested in emulating it, we have the codebase already (it is usually possible only if you're able to access to the video memory, but sometimes workarounds are possible):

ANSI-VT
https://github.com/z88dk/z88dk/tree/master/libsrc/stdio/ansi

For the ANSI VT you should provide at least f_ansi_char and f_ansi_scrollup.  f_ansi_cls and f_ansi_dline can be obtained from generic functions based on f_ansi_char but they'll be slower,  f_ansi_bel is the sound option and it could be left empty or replaced with some sort of visual warning, f_ansi_attr deals with the colors. 
e.g:
https://github.com/z88dk/z88dk/tree/master/libsrc/target/cpc/stdio/ansi
https://github.com/z88dk/z88dk/tree/master/libsrc/target/zx/stdio/ansi
etc..

VT52
The core should be here, but there's probably a better way to understand how to customize it, probably on the wiki.
https://github.com/z88dk/z88dk/blob/master/libsrc/stdio/fputc_cons_generic.inc
See also cpm.cfg, it includes definitions for specific console sizes, etc..

litwr2, 2022-06-11:
Thank you very much for this detailed answer.  But please, help me a bit more.
I haven't still understood how to add a new disc format. :( I can add a new record z88dk/src/appmake/cpm2.c but how to add 10240 bytes of data of boot tracks?  I doubt that it must be typed in a record of the format list in this file.
I am also baffled with idea of making of a target specific library which includes a special fputc_cons version.  It must be a specific library for CP/M.  I  could not find any example for this. :( The LVIV example is not for CP/M.  Please provide me some hint.

suborb, 2022-06-11:
Sorry, just catching up a little.

Have you got a pointer to a disassembly of the bdos? Or even a binary. If the CLS code isn't 0x1a then I'm guessing the move cursor code is different as well. So several tweaks may have to be made.

I'm also wondering whether we should change the code in fputc_cons() to just use the ADM-3a code of 0x1a to clear the screen? Thoughts @zx70? That way we can use a pragma to override for the Corvette and specify it as 0x1f

In the past I've done bootable discs by assembling the boot code as part of the crt: outputting to a separate file and handling that in appmake (see, pc88 for example). Do bootable discs on the Corvette boot into CP/M and then autostart the program?

litwr2, 2022-06-12:
Sorry I don't understand what to do.  I do not need a full terminal emulation now, I just want to implement a basic CP/M console.  IMHO the only problem is CLS, we need to add something to z88dk that allows to remap 0x12 into 0x1f.  Of course the move cursor code is different but it doesn't required for a basic console.  I still can't understand how to make specific libsrc/target/cpm/stdio/fputc_cons.asm for the Corvette. :(
The Corvette uses CP/M 2.2 and some its enhancements.  CP/M 2.2 doesn't autostart any program.  But there are hacks which allow this, is it important for z88dk?

zx70, 2022-06-12:
@suborb, that's not a bad idea. 
It's a bit a pity because the simpler cp/m emulators rely on the OS console,  which is often vt100 capable.

In any case,  if you don't mind,  go for that..  a #pragma option to define a custom control character is better than the current mess.

suborb, 2022-06-12:
Okay, I switched it CP/M over to the above and by default it will clear the screen using the ADM-3a code.

Which means, for the corvette, if you add the option -pragma-define:CLIB_FPUTC_CLS_CODE=0x1f into the appropriate SUBTYPE line you'll be able to clear the screen.

Re the disks - I though you were creating autoboot discs with the code in corvette.c I'm a bit puzzled by it not working without that code being there since I can't see +corvette being referenced by the config file.

litwr2, 2022-06-13:
Thank you very much!  It works, I can properly run ADVENTURE now!
I have just closed my pull request.
Now I am trying to understand how to make the second disc format that will be bootable.  It can take a while.

closed 2022-06-18 by litwr2