z88dk/z88dk #1981: [cpc][udg] Broken CPC games that use UDG but CPCRSLIB is OK

Fabrizio-Caruso, 2022-02-20:
Some CPC libs are broken in the latest snapshots. In particular all my Cross-Lib games crash at start up if compiled with UDG but do work if compiled with CPCRSLIB.

I can reproduce this in Cross-Lib with:
`xl bomber cpc` -> KO with latest Z88DK  (19/02/2022)  but OK with 29/09/2021 snapshot
or
`xl horde cpc` -> KO with latest Z88DK (19/02/2022) but OK with 29/09/2021 snapshot

`xl bomber cpc_cpcrslib` -> OK 

I have included a broken binary and its corresponding .map file.

[BROKEN_build.zip](https://github.com/z88dk/z88dk/files/8104989/BROKEN_build.zip)

suborb, 2022-02-20:
Looking through the CPC code nothing of any consequence has changed for a while,  so I had a look elsewhere and spotted that the stack is being setup incorrectly - it's being set to the 65535 which is a bad place for it to be.

From the. bad one:

```
grep SP XbomberXXX_cpc_mode0.map
DEFINED_REGISTER_SP             = $0001 ; const, local, , cpc_crt0, , C:/cygwin/tmp/zcc000026CCB80D10/zcc_opt.def:11
REGISTER_SP                     = $7FFFFFFF ; const, local, , cpc_crt0, , C:/cygwin/tmp/zcc000026CCB80D10/zcc_opt.def:12
```

From the good one:

```
DEFINED_REGISTER_SP             = $0001 ; const, local, , cpc_crt0, , C:/cygwin/tmp/zcc0000302894F810/zcc_opt.def:11
REGISTER_SP                     = $FFFFFFFF ; const, local, , cpc_crt0, , C:/cygwin/tmp/zcc0000302894F810/zcc_opt.def:12
```

From a build of anything on my machine:

```
DEFINED_REGISTER_SP             = $0001 ; const, local, , cpc_crt0, , /tmp/tmpzccnoo5u3i9/zcc_opt.def:3
REGISTER_SP                     = $FFFFFFFFFFFFFFFF ; const, local, , cpc_crt0, , /tmp/tmpzccnoo5u3i9/zcc_opt.def:4
```

It's looking like a 32/64 bit issue somewhere along the lines. What binaries are you running: the 32 bit windows or 64 bit windows ones?

Fabrizio-Caruso, 2022-02-20:
@suborb 
I am using whatever is in:
http://nightly.z88dk.org/z88dk-win32-20220219-aef53e1d2-19352.zip

I have tested it against a similar snapshot taken from the same page. So one is 32 o 64 bit, the other is probably using the same architecture.

Fabrizio-Caruso, 2022-02-20:
@suborb 
Removing `pragma output REGISTER_SP = -1` seems to fix the issue but was it working before?

suborb, 2022-02-20:
Quite....so I'm trying to figure out how REGISTER_SP=-1 gets written out as a value of 0x7fffffff in the bad case (so a positive number)

My hunch is that a z80asm change has caused this (there was a quite a big change just after Christmas), but only on certain platforms. However, I can't reproduce it on:

- macOS x86_64
- Linux x86_64
- Linux i686
- Win x86_64 (running under wine)
- Win i686 (running under wine)

Could you possible add the --list option and send me the generated .lis file - maybe that will help narrow it down.

Fabrizio-Caruso, 2022-02-20:
I have built with and without the SP pragma with the snapshot from yesterday (19/02/2022).
I have included the .dsk, .lis and .map files.

[NO_SP_PRAGMA_build.zip](https://github.com/z88dk/z88dk/files/8105168/NO_SP_PRAGMA_build.zip)
[WITH_SP_PRAGMA_build.zip](https://github.com/z88dk/z88dk/files/8105169/WITH_SP_PRAGMA_build.zip)

suborb, 2022-02-21:
Well, I wasn't expecting to see this:

```
    10                          IF !DEFINED_REGISTER_SP
    11                                  defc    DEFINED_REGISTER_SP  = 1
    12                                  defc REGISTER_SP  = 0xffffffff
    13                                  IFNDEF REGISTER_SP
    14                                  ENDIF
    15                          ENDIF
```

It's supposed to be printed out in decimal! That might mean the issue is the zcc driver.

suborb, 2022-02-21:
I've now tried it out on a windows vm and I still can't reproduce this.

Fabrizio-Caruso, 2022-02-21:
@suborb What can't you reproduce? 
Are you trying to build the game with Cross-Lib?

suborb, 2022-02-22:
I was trying on the command line, I didn't spot that you were using a zpragma.inc file.

Anyway, that's sorted for the moment, the underlying change that triggered it is unknown to me at the moment though.

commit a201cf3f (2022-02-22): Fix #1981: Some number extension trouble

This code hasn't changed in years, but what consumes possibly has so
write a decimal rather than hex number to be certain.
closed 2022-02-22 by suborb