z88dk/z88dk #1984: zsdcpp: not found when using docker

chris-y, 2022-02-22:
I have a GitHub action which uses the z88dk/z88dk docker image.  I recently pushed an update to my repo which did not change either the C code, the macro or the GitHub action.  This was last run about 17 months ago, when it worked.  Today I get the following:

```
zcc +zxn -v -startup=30 -clib=sdcc_iy -SO3 --opt-code-size tzx2tap.c -o tzx2tap -subtype=dotn -create-app -pragma-define:CLIB_OPT_SCANF=0 -pragma-define:CLIB_OPT_PRINTF=0x5605

PROCESSING tzx2tap.c
zsdcpp -iquote"." -D__Z88DK -D__ZXNEXT -D__Z80 -DZ80 -D__Z80N -D__ESXDOS_DOT_COMMAND -D__NEXTOS_DOT_COMMAND -D__SDCC -D__SDCC_IY -isystem"/opt/z88dk/lib/config//../..//include/_DEVELOPMENT/sdcc"  -D__SDCC  "tzx2tap.c" "/tmp/tmpXXbPnCpa.i2"
sh: zsdcpp: not found
```

I see in https://github.com/z88dk/z88dk/issues/1399 this same issue was reported on MacOS and the workaround was to rename the file.  But the docker should already have that fixed if it is relevant here?

suborb, 2022-02-22:
It looks like there was a change back in July 2020 that accidentally disabled building of sdcc - I think this probably correlates nicely with the timescale.

A new image used to be generated nightly so fixing this up should be easy...

Unfortunately docker made some changes last year that makes using their previously free build infrastructure non-trivial. I raised #1810 at the time and concluded from the feedback that it wasn't really used.

I genuinely don't know how to unblock this issue, so thoughts welcome.

closed 2022-12-16 by feilipu

chris-y, 2022-12-17:
I can't see how this is complete? Still showing as 2 years old on dockerhub.

suborb, 2022-12-17:
> Unfortunately docker made some changes last year that makes using their previously free build infrastructure non-trivial. I raised https://github.com/z88dk/z88dk/issues/1810 at the time and [...]

Given the lack of feedback on #1810 and the download statistics, it's apparent that very few people used the docker hub images and there are more useful things to work on than jumping through dockerhub licensing changes.

As a result of this issue becoming active again, it's obvious that the images hosted at dockerhub are out of date so I've deleted them to avoid confusion.

As was the conclusion on #1810, if a docker installation of z88dk is needed then self build is the only way.

desertkun, 2022-12-18:
It actually would have been very nice to have the thing dockerized since due to lack of an image I had to build an image myself. But since from my experience, alpine is the most popular docker os, having a z88dk package on it would kill the need for the image, because often (like in make case) the docker doesn't needed on it's own, what's needed is to compile something and then dispose of it.

This snippet builds everything and leaves little to no garbage, some packages might not be actually required.

```
FROM alpine:3
RUN apk update && apk add --no-cache cmake git build-base python3 py3-pip py3-setuptools \
    build-base libxml2 libxml2-dev m4 perl bison flex ragel perl-utils perl-dev dos2unix re2c jpeg-dev zlib-dev && \
    cpan install Modern::Perl YAML::Tiny && \
    git clone --depth 1 --recursive https://github.com/z88dk/z88dk.git /z88dk && cd /z88dk \
    && chmod 777 build.sh && ./build.sh && make install && rm -rf /z88dk && \
    apk del cmake git build-base python3 py3-pip py3-setuptools \
        build-base libxml2 libxml2-dev m4 perl bison flex ragel perl-utils perl-dev dos2unix re2c jpeg-dev zlib-dev && \
    rm -rf /var/cache/apk/*
```

suborb, 2022-12-18:
As an update, I've figured out how to add a GitHub action to build and push a docker image so the latest tag at dockerhub should now be the latest master code from the repo.

desertkun, 2022-12-26:
@suborb didn't take much before it broke 😄 
https://github.com/z88dk/z88dk/actions/runs/3783886584/jobs/6432769052
```
re2c: No such file or directory
```

feilipu, 2022-12-27:
Just a CI burp.
It is fine now.