z88dk/z88dk #1992: [sdcc][cywin 64bit] Compilation fails when using -compiler=sdcc under cygwin 64-bit

Fabrizio-Caruso, 2022-03-17:
Hi
I am reporting a problem from a user (@[christopheKohler](https://github.com/christopheKohler)) of Cross-Lib (not me) under Cygwin 64-bit with snapshot: z88dk-win32-20220311-c8d2246b6-19435.zip
Z88DK which works fine except when `-compiler`=sdcc is used.
We have tried a simple helloworld example, which fails with the error shown on the snapshot:
[image]

The helloworld example is just:
```
#include <stdio.h>

int main(void)
{
        printf("hello world\n");

        while(1){};
        return 0;
}
```

feilipu, 2022-03-18:
Untested, but it may be a combination caused by [committing a new ZSDCC](https://github.com/z88dk/z88dk/pull/1969), but possibly not having the new windows binaries built.
The timing of the commit (9 days ago) and the [original issue](https://github.com/Fabrizio-Caruso/CROSS-LIB/issues/282) seems somewhat coincidental.

suborb, 2022-03-19:
It looks like the translation from asz80 to z80asm mnemonics isn't happening.

I'm not sure how this could happen, but I've just updated the sdcc binaries so hopefully that solves it, if it doesn't then adding a `-v` will show us what commands are being invoked.

Fabrizio-Caruso, 2022-03-20:
@suborb I am not sure what has been updated. You mean sdcc (not in Z88DK)? 
So, what should we do now? Would we wait for the next integration of sdcc into Z88DK? Or do you mean that today's snapshot might already have a fix? If yes, I will ask Christophe to test it again. If it fails he could try with the -v option.

christopheKohler, 2022-04-14:
I tested today with latest nighly build ans still same problem.

pauloscustodio, 2022-07-25:
In a Cygwin64 environment, I've tried to extract the z88dk-win32-20220311-c8d2246b6-19435.zip file, configure ```PATH``` and ```ZCCCFG```, and when calling ```zcc +zx81 -v test.c -lndos -create-app``` I get ```Can't open config file /tmp/z88dk/lib/config/zx81.cfg```, although I can can ```cat``` the file.

Then I tried to recompile zcc in the extracted tree and run ```zcc``` again, and the program stops later with another error, now in the ```ucpp``` program:
```
PROCESSING test.c
z88dk-ucpp -iquote"." -DZ80 -DZX81 -D__ZX81__  -DSCCZ80 -DSMALL_C -D__SCCZ80 -I"/cygdrive/c/cygwin64/tmp/z88dk/lib/config/../../\include\arch\zx81" -isystem"/cygdrive/c/cygwin64/tmp/z88dk/lib/config/../..//include"    "test.c" "/tmp/tmpjiD4jlO0.i2"
test.c: line 1: file 'stdio.h' not found
```

Compiling ```ucpp``` in cygwin, it now stops in ```zpragma``` with a similar ```Cannot open``` file errror.

It seems binaries compiled for mingw have trouble to open files when running in a cygwin environment.
Maybe we need to cross-compile for cygwin separately?

suborb, 2022-09-30:
> It seems binaries compiled for mingw have trouble to open files when running in a cygwin environment.

I suspect that the implementation of `open()` used by the mingw builds doesn't understand `/cygdrive/c` - what happens if you set ZCCCFG to the `c:/temp/` (or whatever the Windows path to that directory is)?