z88dk/z88dk #1995: [psg][sms/gamegear][bug?] How to use and compile code to produce sounds through the psg APIs?

Fabrizio-Caruso, 2022-03-30:
I have never managed to figure out how to produce sounds with the PSG APIs of Z88DK for targets such as SMS and GameGear (for which also bit-banging is not working).

If I try to compile:
```
#include <psg.h>

int main(void)
{
    unsigned char i;
    unsigned char j;
    
    psg_init();                       
    psg_volume(0, 15);        
    psg_channels(chan0, chanNone); // set first channel to tone generation, reset noise channels
          
    for(i=0;i<16;i++)
    {
        psg_tone(0, 500-(i*16));
        for(j=0;j<150;++j) {}
    }         
    psg_volume(0, 0);
    return 0;
}
```
with
```
$ zcc.exe +sms generic_psg.c -lndos
```
I get:
```
generic_psg.c::main::0::1:4: error: undefined symbol: psg_channels
  ^---- psg_channels

Compilation aborted
```

suborb, 2022-03-31:
For the sn76489 psglib was imported which supports sound effects. There's an example here: https://github.com/z88dk/z88dk/tree/master/examples/sound/psglib

Fabrizio-Caruso, 2022-03-31:
In my case I would like to produce some simple short beeps, clicks and noise sounds. The code above should work but it doesn't compile.

suborb, 2022-03-31:
I'm nowhere knowledgeable enough about the features of the supported PSG chips, but I believe `psg_channels()` is only supported on the AY since the SN doesn't have an equivalent feature. If you remove the call to `psg_channels` does it work?

Fabrizio-Caruso, 2022-03-31:
@suborb It may be working without it because I hear some sounds on Mednafen. I hear nothing on KegaFusion (which produces sounds on commercial games).
So I wonder what the result is on real hardware. Has anyone tried it?
I have the same results with the bit banging APIs: sounds with Mednafen and no sounds on KegaFusion.

zx70, 2022-03-31:
There's been also some work around the 1bit functions and the sn76489   (games.h - bit_fx(), etc).
In games.inc I see definitions for the following targets:

Sharp MZ-800 + MZ-1500
SC-3000
M5
MTX
SMS
PV2000
RX78
SMC777
PENCIL12

..I understand you weren't able to get it working on the SMS, right ?
Have you had similar problems on other targets in this list as well ?

closed 2022-08-09 by Fabrizio-Caruso