z88dk/z88dk #2001: [psg][tvc] Custom sound HW

dikdom, 2022-04-09:
Hello,

I plan to write some code for a custom soundcard made recently for the TVC (hw). The problem is that the port of the psg chip is not fixed, it depends on which expansion slot is used on TVC. (actually there are two different cards use the psg chip)
IRL it can be runtime detected but as I see there is no way I could easily implement the psg chip support that way into the TVC lib as the port must be prefefined during library compile time...

Any hint? Shall I start write #defines in the psg.h ?

Thanks!

zx70, 2022-04-10:
I'd chose a standard address for the card.  This same problem will affect any other software creation as well.
If you prefer to keep it flexible, then you should, in example,  work on a copy of the psg functions and alter them to get the port assess from a memory location.

suborb, 2022-04-12:
What PSG is it? AY, SN or SAA1099?

If an AY then it's fairly easy to handle the hardware regardless of which slot it's in: all the functions that write to the AY are in the target directory so you could do something like this (you may need two variables depending on whether it's easy to step between the address and data ports):

```
; target/tvc/psg/psg_port.asm
SECTION bss_clib

__psg_port: defw 0

SECTION code_crt_init

; Detect port
ld. (__psg_port),hl
```

```
; set_psg etc
ld bc,(__psg_port)
out (c),a
```

For an SN it would be a bit trickier - I'd need to have a think, likewise for SAA1099.

dikdom, 2022-04-16:
Thanks!
And the TVC-GameCard was not mentioned yet, which also has an SN76489 chip, but with different base clock (3.125MHz instead of the usual 3.579MHz).
It has two extra joy ports which I don't think could fit into the game/joystick() framework.
But no need to rush, I plan to finish the gfx routines first, then some file IO routines shall be implemented.. not in the next days... ;)

closed 2022-06-15 by zx70