z88dk/z88dk #2004: inf loop optimization failure...

dikdom, 2022-04-15:
Hello,
I have a simple program:

```
void main() {
  for(;;) {}
}
```
When I try to compile I got this message:
```
$ zcc +tvc --list -create-app inf.c -o inf.cas
inf.c::main::0::1:1: error: undefined symbol: i_2
  ^---- i_4
inf.c::main::0::1:1: error: undefined symbol: i_4
  ^---- i_2
```
in the list file this can be seen:
```
inf.c:
                                        C_LINE  1,"inf.c"
     1                                  SECTION code_compiler
     1                          
     1                          ; Function main flags 0x00000000 __stdc
     1                          ; void main()
     1                                  C_LINE  1,"inf.c::main::0::1"
inf.c::main::0::1:
     1                          ._main
     1   000000 c30000                  jp      i_2     ;EOS
     1                          .i_3
     1   000003 c9                      ret
     1                          
     1                          
     1                          ; --- Start of Optimiser additions ---
     1                                  defc    i_2 = i_4
     1                                  defc    i_4 = i_2

```
Adding a `-O 0` switch to the compiler would do the trick.
Is it normal that an infinite loop shall not complie with the default compiler options? Or is it me who does something wrong?
Thanks!

feilipu, 2022-04-16:
Interesting. And yet it works correctly with `while(1)`.

```c
void main(void)
{
    while(1)
        {}
}
```

suborb, 2022-04-16:
This will be due to some rules I put in to fold labels together - obviously prune a bit too far.

commit 05664c40 (2022-05-10): Fix #2004: Zap opt rule with incorrect conditions
closed 2022-05-11 by suborb