z88dk/z88dk #2005: TVC pull request

dikdom, 2022-04-22:
Hello,

could you have a look at the TVC pull request I made a week ago? I just would need a feedback on my change request, shall I continue that way or do I need any (serious) modification?

Thanks in advance!

feilipu, 2022-04-22:
There's a few irl issues to manage first, afaik.
Stand by. 👍

suborb, 2022-04-22:
Sorry I've been ill and distracted recently...

I'm happy with the approach, the callee stuff is a bit odd for graphics in general. It tends to work best when the core routine is in assembler and takes parameters in registers - that way the parameters can just be popped off the stack, return address restored and execution continued in the implementation. 

With the macros, the callee function will be called for normal usage so it is worth optimising for it. In the `circle()` case you should be able to rename it to `circle_callee` annotate with `__z88dk_callee` and the compiler will sort it for you.

The `circle()` case would only be called via function pointers (so realistically, quite rarely) so can simply call `circle_callee()` itself.

dikdom, 2022-04-22:
> There's a few irl issues to manage first, afaik. Stand by. +1

Sorry to hear!
Aye captain, I am standing by!

dikdom, 2022-04-22:
> Sorry I've been ill and distracted recently...
> 
> I'm happy with the approach, the callee stuff is a bit odd for graphics in general.  
> [...]

Ok, no worries, I was sick also recently, not a good way spending our time, I know. Hope you're feeling better now!

wrt circle_callee: I had to implement it, because without implementing it explicitely the test file linking (or even the compile of the libsrc?) failed. Anyway, no other `_callee` had to be implemented (despite many of them should be in used, based on the `graphics.h`), somehow those routines are found by the linker, I have no idea how. Could you share some thoughts on this?

dikdom, 2022-04-23:
@suborb 
Erm... something is not ok. I compiled a test file using the new circle/circle_callee and it freezes.

1) I modified the `circle_callee` as you suggested. Now it is a .c file (.asm doesn't exist anymore) and the function header looks like as in the .h file:
```
void __LIB__ circle_callee(int x0, int y0, int radius, int skip) __smallc __z88dk_callee {
```
though it seems to me that it doesn't matter what decorators I provide in the implementation, in the compiler generated .lis file the function's assembly list starts always with
```
    16                          ; Function circle_callee flags 0x000002c0 __smallc __z88dk_callee
    16                          ; void circle_callee(int x0, int y0, int radius, int skip)
    16                          ; parameter 'int skip' at sp+2 size(2)
    16                          ; parameter 'int radius' at sp+4 size(2)
    16                          ; parameter 'int y0' at sp+6 size(2)
    16                          ; parameter 'int x0' at sp+8 size(2)
    16                          	C_LINE	16,"circle_callee.c::circle_callee::0::0"
circle_callee.c::circle_callee::0::0:
    16                          .circle_callee
    16                          	GLOBAL	_circle_callee
    16                          ._circle_callee
    16   000000 210400          	ld	hl,4	;const
    16   000003 cd0000          	call	l_gintspsp	;
    16   000006 210000          	ld	hl,0	;const
    16   000009 e5              	push	hl
    16   00000a e5              	push	hl
```
(even if I provide a -O0 for the lib's make file)

Ok, at the beginning of the function I create 3 local variables, the first one shall get the value of the 3rd argument (SP+4), the other two are initialized to 0. The interesting part (at least for me) is getting this 3rd arg of the method.

2) the `call l_gintspsp` is suspicious for me, so I tried to get its source. I was not able to do so, so I'm just guessing that it is generated somehow from the file `lib/z80rules.8` file, which contains this:

```
        call    l_gintspsp      ;
=
        add     hl,sp   ;l_gintspsp
        ld      a,(hl)
        inc     hl
        ld      h,(hl)
        ld      l,a
        push    hl
```
The heck is that? :) I am still assuming this is an assembly subroutine, so a RET would be generated at the end. Or is it a macro-like structure?? 
Anyhow, it gets 4 in HL, so before that `push hl` the HL contains the 3rd param of `circle_calle`, and it is pushed back?? The SP now points a below the first local param of the circle_callee, returning now would be a mistake.. 
If this is a macro, then it makes sense. But then the .lis file is not really accurate...

3) the circle.c contains now simply this:
```
#include <graphics.h>

void circle(int x0, int y0, int radius, int skip) {
    circle_callee(x0, y0, radius, skip);
}
```
but the generated .lis file contains this:
```
circle.c:
    13                          	C_LINE	15,"circle.c"
    15                          	SECTION	code_compiler
    15                          
    15                          ; Function circle_callee flags 0x000002c0 __smallc __z88dk_callee
    15                          ; void circle_callee(int x0, int y0, int radius, int skip)
    15                          ; parameter 'int skip' at sp+2 size(2)
    15                          ; parameter 'int radius' at sp+4 size(2)
    15                          ; parameter 'int y0' at sp+6 size(2)
    15                          ; parameter 'int x0' at sp+8 size(2)
    15                          	C_LINE	15,"circle.c::circle_callee::0::0"
circle.c::circle_callee::0::0:
    15                          .circle_callee
    15                          	GLOBAL	_circle_callee
    15                          ._circle_callee
    15   000000 210800          	ld	hl,8	;const
    15   000003 cd0000          	call	l_g2intspsp	;
    15   000006 210800          	ld	hl,8	;const
    15   000009 cd0000          	call	l_gintspsp	;
    15   00000c 210800          	ld	hl,8	;const
    15   00000f 39              	add	hl,sp
    15   000010 cd0000          	call	l_gint	;
    15   000013 e5              	push	hl
    15   000014 cd0000          	call	circle_callee
    15   000017 c1              	pop	bc
    15   000018 f1              	pop	af
    15   000019 f1              	pop	af
    15   00001a f1              	pop	af
    15   00001b f1              	pop	af
    15   00001c c5              	push	bc
    15   00001d c9              	ret
```
Why on Earth is the `circle` function labelled with `circle_callee` ??? I know `circle()` is not used, but still. What is going on, what do I miss here??
Does the `circle.c` generated .o file overloads/overwrites the `circle_callee()` library function from `circle_callee.c`??

(it was working when the callee simply called the circle c routine... :) )

suborb, 2022-04-28:
If you've not figured out, what's happening is a common gotcha.

graphics.h has this:

```
#define func(x,y) func_callee(x,y)
```

Which means that when you compile your C file:

```
void func(int x, int y) {
```

it ends up being changed into:

```
void func_callee(int x, int y) {
```

I think this tripped Stefano recently but I can't find the reference: the simplest resolution is just to add `#undef func` after inclusion of `graphics.h`

dikdom, 2022-05-03:
Thanks! It did the trick.

closed 2022-05-03 by dikdom